espressif/arduino-esp32 #5726: Build fails with undefined references to MD5Builder::   MD5Init, MD5Update, MD5Final

KD4Z, 2021-10-01:
Board:							ESP32 -WROOM-32 Dev Module 
Core Installation version:			2.0.0 release
IDE name:						Visual Studio Code / bash terminal
Flash Frequency:					80Mhz
PSRAM enabled:                                     yes
Upload Speed:					115200
Computer OS:                                        Windows 10

ArduinoCore 2.0.0
IDF 4.3.1 Release tag
Compiler version: 8.4.0

Attempting to get clean build after updating to IDF 4.3.1 and ArduinoCore v2.0.0.    All components compile, but linking fails.
Also tried current IDF 4.3.1 master branch, no help.  Not sure what IDF version to use.  Description of ArduinoCore 2.0.0 mentions "latest development version of IDF",

Did I miss changing something in menuconfig?

```
Toolchain path: /home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2021r1
Compiler version: 8.4.0
Python requirements from /mnt/c/git/esp/esp-idf/requirements.txt are satisfied.

Arduino libraries in build: ArduinoOTA AsyncUDP BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS FFat FS HTTPClient HTTPUpdate HTTPUpdateServer LittleFS NetBIOS Preferences README.md RainMaker SD SD_MMC SPI SPIFFS SimpleBLE Ticker USB Update WebServer WiFi WiFiClientSecure WiFiProv Wire
Project is not inside a git repository, or git repository has no commits
will not use 'git describe' to determine PROJECT_VER.
App "slserver" version: 1
LD build/slserver.elf
/home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/c/git/esp/slserver/build/ArduinoCore/libArduinoCore.a(MD5Builder.o):(.literal._ZN10MD5Builder5beginEv+0x0): undefined reference to `MD5Init'
/home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/c/git/esp/slserver/build/ArduinoCore/libArduinoCore.a(MD5Builder.o):(.literal._ZN10MD5Builder3addEPht+0x0): undefined reference to `MD5Update'
/home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/c/git/esp/slserver/build/ArduinoCore/libArduinoCore.a(MD5Builder.o):(.literal._ZN10MD5Builder9calculateEv+0x0): undefined reference to `MD5Final'
/home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/c/git/esp/slserver/build/ArduinoCore/libArduinoCore.a(MD5Builder.o): in function `MD5Builder::begin()':
/mnt/c/git/esp/slserver/components/ArduinoCore/cores/esp32/MD5Builder.cpp:32: undefined reference to `MD5Init'
/home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/c/git/esp/slserver/build/ArduinoCore/libArduinoCore.a(MD5Builder.o): in function `MD5Builder::add(unsigned char*, unsigned short)':
/mnt/c/git/esp/slserver/components/ArduinoCore/cores/esp32/MD5Builder.cpp:37: undefined reference to `MD5Update'
/home/kd4z/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /mnt/c/git/esp/slserver/build/ArduinoCore/libArduinoCore.a(MD5Builder.o): in function `MD5Builder::calculate()':
/mnt/c/git/esp/slserver/components/ArduinoCore/cores/esp32/MD5Builder.cpp:97: undefined reference to `MD5Final'
collect2: error: ld returned 1 exit status
/mnt/c/git/esp/esp-idf/make/project.mk:574: recipe for target '/mnt/c/git/esp/slserver/build/slserver.elf' failed
make: *** [/mnt/c/git/esp/slserver/build/slserver.elf] Error 1

```

me-no-dev, 2021-10-02:
use IDF Master (IDF4.4)

closed 2021-10-02 by me-no-dev

KD4Z, 2021-10-02:
So, that is what I had already tried.    To be sure, I nuked the whole tree again, and re-cloned the repository just now ,  I don't see any 4.x anything tags or other branches mentioning 4.anything.  So I built against the master branch.    Same errors listed in OP.  No others other than warnings.

Am I still using the wrong tag/branch?   I must have something else configured wrong, but what?

```
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
Here is the last commit...

```
$ git log
commit 3e370c4296247b349aa3b9a0076c05b9946d47dc (HEAD -> master, origin/master, origin/HEAD)
Merge: 8d74bf87ec 0f7b90d205
Author: morris <maoshengrong@espressif.com>
Date:   Fri Sep 24 16:12:24 2021 +0000

    Merge branch 'bugfix/filter_out_readme_searching_components' into 'master'

    cmake: allow to have README file in the components folder

    See merge request espressif/esp-idf!15330
```