espressif/arduino-esp32 #5727: ESP32-S2 USB CDC On Boot - Serial input drops incoming data sent quickly from the PC

Jasoroony, 2021-10-02:
### Hardware:
Board:							ESP32-S2 Saola 1R Dev Kit featuring ESP32-S2 WROVER
Core Installation version:			https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
IDE name:						Arduino IDE 1.8.15
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 7 x64

### Description:
When using the "USB CDC On Boot = Enabled" Serial port, it will lose incoming characters (received by the ESP32-S2) unless the PC slows down the sent data by using smaller writes with delays between each write.  The example sketch below sets the receive buffer to 2K, however even if I only write 1024 bytes in one Windows API call to WriteFile() I will often only see ~800 bytes arrive.

The sketch runs a tight loop reading and throwing away characters, but it keeps a count.  Once there is a delay of 2 seconds when the incoming data stops, it reports how many characters it saw.  Of course, this is a bare minimum sketch, which shows the larger problem I'm having in my project.

To try the sketch, configure the USB-COM port for the device to be COM1 (single digit COM ports can be used from the command line) and open the device from PuTTY.  Type a few characters, wait a couple of seconds, it will report back how many characters were typed.

The device is now primed (DTR & RTS has been configured by PuTTY) where you can close PuTTY but get it ready to open again (have the PuTTY config window open and ready to connect)...  In a DOS console (command prompt), type "copy my1024b.bin \b com1" (the file needs to be created beforehand and should have 1024 bytes in it)...  Once it says "1 file(s) copied.", quickly press the Open button in PuTTY and it will tell you how many characters it got.

### Sketch:
```cpp

static int c = 0;
static unsigned long tm;
static char buf [260];

void setup () {

    delay (2500);
    Serial.begin (921600);
    Serial.setRxBufferSize (2048);
}

void loop () {

    if (Serial.available ()) {
        Serial.read ();
        c++;
        tm = micros ();
    }
    if ((c) && (micros () - tm >= 2000000)) { // 2s
        sprintf (buf, "%05d\n", c);
        Serial.write (buf);
        c = 0;
    }
}
```

### Debug Messages:
```
00825
```

me-no-dev, 2021-10-02:
try calling `Serial.setRxBufferSize` before `Serial.begin`

Jasoroony, 2021-10-02:
I tried that, but it didn't help.  I got 00412 as output from a file with 1024 bytes in it.

I made a typo in the copy command earlier (I typed the wrong slash) this is the command to use: "copy my1024b.bin /b com1" but depending on what's in the file, you don't really need the "binary" flag.

Also, just for completeness, changing the COM port number is in "Computer Management"->"Device Manager"->"Ports COM & LPT" double-click your device, select the "Port Settings" tab and press the "Advanced" button.  COM Port Number is at the bottom.

chegewara, 2021-10-02:
Is there any reason you read 1 byte only with every read call?

Jasoroony, 2021-10-03:
Worth a try, right (it's not how my project does it, but I was keeping the example simple):

```
static int c = 0;
static unsigned long tm;
static char buf [260];

void setup () {

    delay (2500);
    Serial.setRxBufferSize (2048);
    Serial.begin (921600);
}

void loop () {
    int i;
    int j;

    i = Serial.available ();
    if (i) {
        while (i) {
            if (i > 260)
                j = 260;
            else
                j = i;
            Serial.read (buf, j);
            i -= j;
            c += j;
        }
        tm = micros ();
    }
    if ((c) && (micros () - tm >= 4000000)) { // 4s
        sprintf (buf, "%05d\n", c);
        Serial.write (buf);
        c = 0;
    }
}
```

Output:

00694

me-no-dev, 2021-10-21:
you can just not enable CDC on boot and do it in the sketch instead.

Jasoroony, 2022-01-19:
Fix locked mutex when fifo is full: https://github.com/hathach/tinyusb/pull/1286

kerryland, 2023-02-06:
> I said that first, me-no-polite :)
> 
> Look for my review of this product at: https://core-electronics.com.au/esp32-s2-saola-1r-dev-kit-featuring-esp32-s2-wrover.html

Interesting there are now no reviews listed. Perhaps the trust-worthy retailer doesn't like the negative ones?

XZCE, 2023-05-01:
It's still "in the system" via the search, but the jerk has hidden it within the product page:

[image]

PR #6413 (2022-03-28): Fixes USB CDC setRxBufferSize(), begin(), _onRX()
## Summary
this PR fixes these issues:
1- ```USBCDC::setRxBufferSize()``` doesn't change RX Queue Size 
2- ```USBCDC::begin()``` always force 256 byte as RX Queue Size
3- ```USBCDC::_onRX()``` drops data when RX Queue Size is full
4- ```USBCDC::_onRX()``` implies in Arduino Task starvation when receiving more bytes than RX Queue can handle
5- ```USBCDC::end()``` resets the board when executed twice in sequence

## Impact
This PR allows ```USBCDC::setRxBufferSize()``` to be called either, before or after, ```USBCDC::begin()```.
It will work in a [... cut]