espressif/arduino-esp32 #5728: 'pthread_mutex_timedlock' was not declared in this scope

mjlev314, 2021-10-03:
I've updated my Board Support from 1.06 to 2.00.

I'm now getting a compilation error 'pthread_mutex_timedlock' was not declared in this scope.

If I go back down to 1.06 the same source code builds correctly.

the exact error is:

c:\users\michael\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:781:12: error: 'pthread_mutex_timedlock' was not declared in this scope

     return __gthrw_(pthread_mutex_timedlock) (__mutex, __abs_timeout);

            ^~~~~~~~

c:\users\michael\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\include\c++\8.4.0\xtensa-esp32-elf\no-rtti\bits\gthr-default.h:781:12: note: suggested alternative: 'pthread_mutex_trylock'

I'm trying to build applications that worked on 1.06 against 2.00.

I'm building this on Windows 10/11 machines with Arduino 1.8.16.

### Hardware:
Board:							Node32S
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     ?no? ?yes?
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
Compilation fails with 2.00 Board support but works with 1.06

Using BLE_Scan.ino example, that same occurs:

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 5; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}

me-no-dev, 2021-10-03:
No issues compiling that example here. @igrr any clues?
[image]

VeniceInventors, 2021-12-03:
Same issue here, but I resolved it by changing the board from:

Tools->Board->**ESP32 Arduino (in sketchbook)**->WEMOS LOLIN32
to
Tools->Board->ESP32 Arduino->WEMOS LOLIN32

I don't know what the difference is but the issue seems to be related to the board selected

VojtechBartoska, 2022-04-07:
@mjlev314 is this till valid?

VojtechBartoska, 2022-04-21:
Hello,

as there was no answer in more than 14 days, I'm closing the issue as expired to keep our backlog manageable.

If it's still needed, please reopen the issue.

Thanks for understanding.

closed 2022-04-21 by VojtechBartoska

hpwit, 2022-11-29:
I am sorry to reopen this topic
But I Have the same issue since I have upgraded to 2.0.4
the strangest thing is that is I complied with memo midi board no issue if I compile with most of the other boards then I have this error.
This is rather annoying I have to admit
Thank you

Degenknauf, 2023-03-15:
Hi there, 
I've struggled with the problem described above nearly one week. The solution is simply to switch to another board. In my case from ESP32 Dev Module to WEMOS Lolin32. Thanks for the hint, which I found so late unfortunately.
Thanks for your work.

klkl140, 2023-11-16:
Changing the board does help me too. Many thanks

mjsir911, 2024-02-26:
Seeing the same issue, @VojtechBartoska can this one be reopened or should I open another one?

VojtechBartoska, 2024-02-26:
@mjsir911 Please open new issue