espressif/arduino-esp32 #5729: Errors and warning compiling the same sketch with the new ESP32 v2.0.0

pberna67, 2021-10-03:
Dear developer

I've noticed errors and warnings compiling the attached sketch using
**Espressif System Package 2.0.0-rc2 or 
Espressif System Package 2.0.0**

But no errors nor warnings emerges when the 
Espressif System Package 1.0.5-rc5 or below versions  are used

The error found bu the compiling the sketch with the  System package Package 2.0.0-rc2 are the following

Make your question, not a Statement, inclusive. Include all pertinent information:
In file included from C:\Users\pabe\Documents\Arduino\MyExamples\MySimpleMQTT\TemperatureMonitor8MQTTClient\TemperatureMonitor8MQTTClient.ino:12:
C:\Users\pabe\Documents\Arduino\libraries\ESP32TimerInterrupt\src/ESP32TimerInterrupt.h:56:4: warning: #warning Using ESP32TimerInterrupt Library [-Wcpp]
   #warning Using ESP32TimerInterrupt Library
    ^~~~~~~
In file included from C:\Users\pabe\Documents\Arduino\libraries\OneWire\OneWire.cpp:144:
C:\Users\pabe\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeInput(uint32_t)':
C:\Users\pabe\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:161:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
C:\Users\pabe\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:161:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
C:\Users\pabe\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
C:\Users\pabe\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:26: error: 'rtc_gpio_desc' was not declared in this scope
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
C:\Users\pabe\Documents\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:26: note: suggested alternative: 'rtc_io_desc'
         uint32_t rtc_reg(rtc_gpio_desc[pin].reg);
                          ^~~~~~~~~~~~~
                          rtc_io_desc
Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\pabe\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.0-rc2\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.51.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board u-blox NINA-W10 series (ESP32).

---------------------------------------------------------------------------------------------------------------------
Describe your system( Hardware, computer, O/S, core version, environment).
PC Windows 10

me-no-dev, 2021-10-04:
This is a problem with the OneWire library. They access the GPIO on a level lower than Arduino.

closed 2021-10-04 by me-no-dev

pberna67, 2021-10-06:
It seems that there is also a problem in WiFi lib