espressif/arduino-esp32 #5730: Bootloader Problem

k3rn3lp4n1c-1337, 2021-10-04:
**HARDWARE**
ESP32C3-Devkit
IDE: Arduino
Toolchain: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
OS: Linux
**Description:**
I have written a firmware updater for the production mode. 
I also compiled a new bootloader with the IDF framework 4.4 with the Fall Back settings "CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE".
The bootloader and the firmware run without problems.
I can also trigger a rollback using the function "esp_ota_mark_app_invalid_rollback_and_reboot()", this also works without problems.
But now that an update has been performed and the new firmware is running for the first time on the OTA_0 partition, the bootloader is set to "UNKNOW" and not "ESP_OTA_IMG_PENDING_VERIFY" as expected, so the app runs in test mode for the first run.
Thus, it is also not possible for me with the test firmware after 10 seconds a provoked restart of the Esp makes a rollback as it was actually intended.

Code Snipped to read image and boot information:
```
Serial.println(F("++++++++++"));
	Serial.println(F("BOOT INFOS"));
	Serial.println(F("++++++++++\n"));

	const esp_partition_t *running = esp_ota_get_running_partition();
	esp_ota_img_states_t ota_state;
	esp_ota_get_state_partition(running, &ota_state);
	EspClass esp;
	char formatStringBuffer[50] = {0};
	sprintf(formatStringBuffer, "Firmware: %s \n", FIRMWARE_VERSION);
	printf("%s", formatStringBuffer);
	sprintf(formatStringBuffer, "HW Version: %s \n", HW_VERSION);
	printf("%s", formatStringBuffer);
	sprintf(formatStringBuffer, "Boot Partition Name: %s \n", bootInfo->label);
	printf("%s", formatStringBuffer);
	sprintf(formatStringBuffer, "Start Address Offset: %02x \n", bootInfo->address);
	printf("%s", formatStringBuffer);
	sprintf(formatStringBuffer, "Start Size: %02x \n", bootInfo->size);
	printf("%s", formatStringBuffer);
	sprintf(formatStringBuffer, "App Size: %i \n", esp.getSketchSize());
	printf("%s", formatStringBuffer);
	if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
		printf("ota state: ESP_OTA_IMG_PENDING_VERIFY\n");
	}else if (ota_state == ESP_OTA_IMG_INVALID){
		printf("ota state: ESP_OTA_IMG_INVALID\n");
	}else if (ota_state == ESP_OTA_IMG_VALID){
		printf("ota state: ESP_OTA_IMG_VALID\n");
	}else if (ota_state == ESP_OTA_IMG_NEW){
		printf("ota state: ESP_OTA_IMG_NEW\n");
	}else{
		printf("ota state: UNKNOW\n");
	}
	if (esp_ota_check_rollback_is_possible() == true){
		printf("Rollback possible: true\n");
	}else{
		printf("Rollback possible: false\n");
	}
	printf("\n");
```
Shouldn't the bootloader be told that the firmware is now in the new partition and change the ota flag to "ESP_OTA_IMG_PENDING_VERIFY" when you run the following code after an update has been loaded onto a partition?

```
if (ESP_OK == esp_ota_set_boot_partition(freeOtaPartition)) {
setNewRunningPartitionForBootloaderFallbackCheck();
esp_ota_end(otaHandler);
delay(2000);
esp_restart();
}
```
I compile the code of the firmware with the Arduino Esp32 2.0.0 Sdk.
I hope someone knows a solution to the problem.

me-no-dev, 2021-10-04:
Try using `esp_ota_set_boot_partition(partition);`. You are compiling with Arduino, but you are not using Arduino API. You are in reality compiling with a recent ESP-IDF master (4.4). Maybe use the Update class instead: https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Update.h

closed 2021-10-04 by me-no-dev

k3rn3lp4n1c-1337, 2021-10-04:
HI i also tried the arduino variant you described above.

This uses like me also the "esp_ota_set_boot_partition(partition)" function

Update.cpp
```
bool UpdateClass::_verifyEnd() {
    if(_command == U_FLASH) {
        if(!_enablePartition(_partition) || !_partitionIsBootable(_partition)) {
            _abort(UPDATE_ERROR_READ);
            return false;
        }

        if(esp_ota_set_boot_partition(_partition)){
            _abort(UPDATE_ERROR_ACTIVATE);
            return false;
        }
        _reset();
        return true;
    } else if(_command == U_SPIFFS) {
        _reset();
        return true;
    }
    return false;
}
```
The Bootloader runs in the same problem:
Boot Partition Name: ota_0 
Start Address Offset: 20000 
App Size: 561632 
ota state: UNKNOW <=========================
Rollback possible: false

Can it be that the created compilat is incompatible with the bootloader?