espressif/arduino-esp32 #5735: Can't install v2.0.0 on arm64

urish, 2021-10-05:
### Hardware:
Board:							any
Core Installation version:			2.0.0
IDE name:						Arduino CLI
Computer OS:                                         Linux-aarch64 (arm64)

### Description:
Installation fails with an error:

> Error during install: Error downloading tool esp32:riscv32-esp-elf-gcc@gcc8_4_0-esp-2021r1: no versions available for the current OS%!(EXTRA *cores.ToolRelease=esp32:riscv32-esp-elf-gcc@gcc8_4_0-esp-2021r1)

Seems like arm64 support was originally added by #4117, but broke due to the introduction of the riscv32 core support.

igrr, 2021-10-05:
Linking https://github.com/espressif/esp-idf/issues/6432 which tracks adding aarch64 host support in ESP-IDF. Once that's released in IDF (planned for esp-2021r2 toolchain release), we can pick these tool releases in arduino-esp32.

It looks like https://github.com/espressif/arduino-esp32/pull/4117 simply added existing 32-bit ARM (`armel`) toolchains as ones to install for aarch64, so I'm not quite sure how that worked. Perhaps one can `sudo dpkg --add-architecture armhf` to install make these 32-bit tools work on a 64-bit system? If that does work, we can try to re-add these `armel` versions of tools.

Jason2866, 2021-10-05:
@igrr Is it planned to release the toolchains for Apple Silicon M1 too?

igrr, 2021-10-07:
@urish correct, installing architecture support is likely not sufficient. At least these `armel` tools need to be listed as downloads for `arm64`, this has to be done in package.json of the arduino-esp32 release. It is possible that other steps will be necessary, e.g. installing libc or other system libraries for `armhf`.

You can download the armel toolchain to your arm64 system (link [here](https://github.com/espressif/arduino-esp32/blob/bd54ee442b923a194ed3b915ed11d420ced17390/package/package_esp32_index.template.json#L83)) and then try to run it. If that fails, check if some additional libraries need to be installed (use `ldd` to find the dependencies). If all that is successful somehow, then we can update package_esp32_index.template.json and include this change into the next release.

igrr, 2021-10-07:
@Jason2866 Yes, https://github.com/espressif/esp-idf/issues/6113 is the tracking issue for M1 toolchain. However it's not going to be part of the next esp-2021r2 toolchain release, maybe will get done in the one after that.

urish, 2021-10-08:
Thanks for the direction! I have changed the [generated board file](https://github.com/espressif/arduino-esp32/blob/gh-pages/package_esp32_index.json) and added the `armhf` binaries for the aarch64-linux-gnu architecture, then installed the new package and voila - it works like a charm!

I also verified I could compile for the following boards:
- esp32:esp32:esp32doit-devkit-v1
- esp32:esp32:franzininho_wifi_esp32s2
- esp32:esp32:esp32c3

all compiled well. This is what the [modified file](https://gist.github.com/urish/0ab8241cc85bd44b847503c51da3d9aa) looks like.

Should go ahead and send a pull request?

igrr, 2021-10-29:
@me-no-dev Do you think we can still add this to 2.0.1?

me-no-dev, 2021-10-30:
yes. will update the toolchains for 2.0.1 :)

urish, 2021-11-10:
Fixed in 2.0.1 and verified on Wokwi. 

Thanks for fixing this! Now the users can enjoy [ESP32-S2 Blink on the Arduino Core](https://wokwi.com/arduino/projects/314916550651937346) :wink:

closed 2021-11-10 by urish