espressif/arduino-esp32 #5737: Modules with PSRAM take long to boot

mcer12, 2021-10-06:
Seems like PSRAM is tested turing boot, which can be disabled in menuconfonfig ("Run memory test on SPI RAM initialization" option). It would be nice to be able to disable the PSRAM test in arduino environment as well.

It is mentioned here:
https://esp32.com/viewtopic.php?t=9448

mcer12, 2021-12-01:
Any news on this? I know 500ms doesn't seem like a big deal but on battery powered devices, it can mean weeks of extra deep sleep time :)

SuGlider, 2022-01-13:
@mcer12 
Sorry for the delay. I already have a solution for it.
Using ```BYPASS_SPIRAM_TEST(bool);``` will allow you to bypass PSRAM test and make the WROVER boot fast.

Example:

``` dart
BYPASS_SPIRAM_TEST(true);

void setup() {
  Serial.begin(115200);
  Serial.println("\n Testing SPIRAM test bypass!");
  Serial.println("Check how long does it take to boot in a WROVER module.");
}

void loop() {
  static uint8_t count = 0;
  Serial.print("-");
  delay(500);
  if (++count == 32) {
    count = 0;
    Serial.println();
  }
}
```

SuGlider, 2022-01-13:
The PR #6135  will solve your need.

closed 2022-01-17 by me-no-dev

ashald, 2022-05-03:
How do I use it? I'm getting an error when I'm trying to add it as in the comment above:
```
src/main.cpp:32:19: error: expected constructor, destructor, or type conversion before '(' token
 BYPASS_SPIRAM_TEST(true);
```

PR #6135 (2022-01-17): Allows user to bypass PSRAM test and boot faster with WROVER
## Summary
Users have requested a way to make the WROVER module to boot faster.
It takes about 1 second to boot in a 4MB PSRAM configuration because of SPI RAM Test routine.

This PR adds a function to ESP32 Arduino that allows the user to programmatically tell Arduino not to spend time on PSRAM test. The default behavior is to test it.

The API is ```BYPASS_SPIRAM_TEST(bool)```

Example:

``` dart
BYPASS_SPIRAM_TEST(true);  // not declaring it or using BYPASS_SPIRAM_TEST(false) will run SPI RAM Test

void setup() {
  Serial.begin(11 [... cut]
commit 1bbe61ab (2022-01-17): Allows user to bypass PSRAM test and boot faster with WROVER (#6135)

Fixes #5737