espressif/arduino-esp32 #5738: FreeRTOS - file or library not found

Lazarus-Rising, 2021-10-06:
Hi, I'm having difficulties compiling Grbl Esp32/bdring onto a Wemos D1R32.
I've followed various 'methods' on Youtube and keep getting errors, the one holding things up now is the one regarding FreeRTOS:
ResolveLibrary:(FreeRTOS.h)1_Esp32-main\Grbl_Esp32\src\I2SOut.cpp:54:22:  fatal  error: FreeRTOS.h No such file or directory.

I've created and downloaded my own FreeRTOS.h files and placed them in the folder along with Config.h etc but keep getting this error.
Can someone steer me right please?
----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|ESP32-S2-Saola|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>|Arduino IDE|Atom + Platform.io|IDF component|VSCode|
|<B>Flash Frequency</B>|40Mhz|80Mhz|
|<B>PSRAM enabled</B>|yes|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|Mac OSX|Ubuntu|

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

chegewara, 2021-10-06:
This is question or issue report to library owner/maintainer:
`ResolveLibrary:(FreeRTOS.h)1_Esp32-main\Grbl_Esp32\src\I2SOut.cpp:54:22: fatal error: FreeRTOS.h`

I would suggest to checkif includes are correct, like this:
`#include "freertos/FreeRTOS.h"`
Although i think that @me-no-dev has been working on change it and allow to use `#include "FreeRTOS.h"`

chegewara, 2021-10-06:
Maybe it is similar to this library, bad file naming, and should be in library:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/FreeRTOS.h

me-no-dev, 2021-10-08:
Please take this to Grbl_Esp32's repo. This is not an issue with Arduino.

closed 2021-10-08 by me-no-dev

bmwhitetx, 2022-01-07:
I got the same error when I had to reinstall GRBL ESP32 on my ESP32 dev module. I fixed it by downgrading the ESP32 Boards. I use the Arduino IDE and went to Boards manager selected ESP32 and installed version 1.0.6. I had updated to 2.0.1 a few months ago. I guess the 2.0 version added the RTOS requirement.

kcb75us, 2022-10-07:
bmwhitetx...
are you able to use bluetooth on your esp32dev? i have a generic esp32 dev also and i cant get the bluetooth arduino library to load.

bmwhitetx, 2022-10-07:
I don’t have a need for Bluetooth so haven’t tried. Sorry.