espressif/arduino-esp32 #5771: Compile errors for a2dp_sink with esp32-s3

pschatzmann, 2021-10-16:
The documentation for the ESP32-S3 is indicating that A2DP should be working.
Compiling your a2dp_sink example gives the following compile errors

 ```
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c: In function 'hci_layer_init_env':
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:184:5: error: implicit declaration of function 'btsnd_hcic_ble_sync_sem_init'; did you mean 'btsnd_hcic_ble_set_phy'? [-Werror=implicit-function-declaration]
     btsnd_hcic_ble_sync_sem_init();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     btsnd_hcic_ble_set_phy
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c: In function 'hci_layer_deinit_env':
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:207:5: error: implicit declaration of function 'btsnd_hcic_ble_sync_sem_deinit'; did you mean 'btsnd_hcic_ble_clear_adv_set'? [-Werror=implicit-function-declaration]
     btsnd_hcic_ble_sync_sem_deinit();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     btsnd_hcic_ble_clear_adv_set
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c: In function 'filter_incoming_event':
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:443:36: error: implicit declaration of function 'btsnd_hcic_ble_get_sync_info'; did you mean 'btsnd_hcic_ble_ext_scan_enable'? [-Werror=implicit-function-declaration]
             BlE_SYNC *sync_info =  btsnd_hcic_ble_get_sync_info();
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                    btsnd_hcic_ble_ext_scan_enable
/Users/pschatzmann/esp/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:443:36: warning: initialization of 'BlE_SYNC *' {aka 'struct <anonymous> *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
cc1: some warnings being treated as errors
 ```

pschatzmann, 2021-10-16:
see https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/bluetooth/classic_bt.html

lbernstone, 2021-10-16:
esp32-s3 is not yet supported in arduino, afaik
esp32-c3 has no support for classic BT; it has BLE only

pschatzmann, 2021-10-16:
I just noticed that I was filing this issue in the wrong forum. I will move it to the ESP32 one.

pschatzmann, 2021-10-16:
https://github.com/espressif/esp-adf/issues/702

lbernstone, 2021-10-16:
I'm happy to answer it for you.  The API docs should not include bluetooth for the C3.  The [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf) is clearer that only BLE is supported.

JayShoe, 2021-10-16:
I'm curious. Does/will the s3 support Bluetooth classic?

lbernstone, 2021-10-16:
No, only the ESP32 will be supporting classic BT.
https://products.espressif.com/#/product-comparison