espressif/arduino-esp32 #5798: ESP32-CAM boot only in download mode

kevin-doolaeghe, 2021-10-23:
I have a problem with my AI THINKER ESP32-CAM: it boots constantly in download mode. Conclusion is that I can upload code to the board but I can't run the program.

Indeed, if GPIO0 is shorted with GND, it boots with code 0x3 (= download mode) but if I remove the short, it continues booting in download mode unlike described in the following article:

See "GPIO 0" section of [this link](https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout)

I think I have a hardware problem because I mesured 0V instead of the normal 3.3V through the 10kOhm pull-up resistance (according to the schematic).

Has anyone already had this problem?
How can I solve it?

Can I just short GPIO0 pin with the 3.3V source?

Thanks in advance,
Kevin

closed 2021-11-14 by SuGlider

mdepa91, 2023-04-05:
for anyone having this problem: the GPIO is connected to the CAM, I'm having issues until the CAM is disconnected. Or try restarting, it should start at some point

dradomtsev, 2025-04-29:
for anyone having this problem: if you have inserted SD-card - remove it before flashing the sketch, it will help