espressif/arduino-esp32 #5822: analogRead within an ISR causes "Interrupt wdt timeout on CPU1"

colonelwatch, 2021-10-30:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation version:			1.0.5 or 1.0.6 (no issue on 1.0.4)
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                     No
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
The below minimal sketch crashes with the same error that I encountered in one of my projects. I have confirmed this crash does not exist when the code is compiled with core version 1.0.4 but does exist with 1.0.5 and 1.0.6.

The core motivation behind this piece of code was a sampling interrupt, which I needed for a project dependent on collecting audio in realtime. Naturally, the crux of the issue seems to be `analogRead(36)` in the ISR because removing it lets the code run normally. For now, my workaround has been to downgrade to 1.0.4.

### Sketch:
```cpp
hw_timer_t *timer = NULL;
void IRAM_ATTR onTimer(){
    analogRead(36);
}

void setup() {
    Serial.begin(115200);

    timer = timerBegin(1, 80, true);
    timerAttachInterrupt(timer, &onTimer, true);
    timerAlarmWrite(timer, 22, true); // approximately 44100 kHz
    timerAlarmEnable(timer);
}

void loop() {
    static unsigned long past = millis();
    if(millis()-past > 2000){
        past = millis();
        Serial.println("All good!");
    }
}
```

### Debug Messages:
The above sketch will crash **many different backtraces**, but here is one of them.

```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40086152  PS      : 0x00060834  A0      : 0x80083711  A1      : 0x3ffbea60  
A2      : 0x3ffb84a0  A3      : 0x3ffbea8c  A4      : 0x00000000  A5      : 0x3ffb84ec  
A6      : 0x00060821  A7      : 0x00000000  A8      : 0x8008643c  A9      : 0x3ffbea60  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060821  A13     : 0x00000001  
A14     : 0x00060823  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400d104c  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086152

ELF file SHA256: 0000000000000000

Backtrace: 0x40086152:0x3ffbea60 0x4008370e:0x3ffbea80 0x400837ed:0x3ffbeab0 0x400e47f0:0x3ffbead0 0x400e4901:0x3ffbeb00 0x400d1123:0x3ffbeb30 0x40080ed5:0x3ffbeb60 0x400810fd:0x3ffbeb80 0x40083ba5:0x3ffbeba0 0x40081105:0x3ffb1f80 0x400d236a:0x3ffb1fb0 0x400864e5:0x3ffb1fd0

Core 0 register dump:
PC      : 0x400eb45e  PS      : 0x00060b34  A0      : 0x800e1f5a  A1      : 0x3ffbc130  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060d20  A7      : 0x00000000  A8      : 0x800e1c4e  A9      : 0x3ffbc100  
A10     : 0x00000000  A11     : 0x40083584  A12     : 0x00060d20  A13     : 0x3ffbb950  
A14     : 0x3ffc01e0  A15     : 0x3ffc01e4  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400eb45e:0x3ffbc130 0x400e1f57:0x3ffbc150 0x4008797a:0x3ffbc170 0x400864e5:0x3ffbc190

Rebooting...
```

And the decoding is the following.

```
PC: 0x40086152: xQueueGiveFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1299
EXCVADDR: 0x00000000

Decoding stack results
0x40086152: xQueueGiveFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1299
0x4008370e: lock_release_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 199
0x400837ed: _lock_release at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 213
0x400e47f0: adc1_lock_release at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c line 1560
0x400e4901: adc1_get_raw at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c line 1582
0x400d1123: __analogRead at C:\Users\knypn\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-adc.c line 140
0x40080ed5: onTimer() at C:\Users\knypn\Documents\PROJECTS\SKETCHPAD\esp32-sampling-issue/esp32-sampling-issue.ino line 3
0x400810fd: __timerISR at C:\Users\knypn\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-timer.c line 88
0x40081105: __timerISR at C:\Users\knypn\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-timer.c line 86
0x400d236a: loopTask(void*) at C:\Users\knypn\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 18
0x400864e5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

me-no-dev, 2021-10-30:
on ESP32 you can not read adc in the interrupt. Set a flag that it was triggered and read the adc in another thread or in the loop.

me-no-dev, 2021-10-30:
to clarify, the driver has changed since 1.0.4 and is now using ESP-IDF API to access the ADC. That API is not usable in ISRs

closed 2021-10-30 by me-no-dev