espressif/arduino-esp32 #5823: error: 'PCNT' was not declared in this scope (ESP32 release v2.0.1RC1)

whogarden, 2021-10-31:
### Hardware:
ESP-WROOM-32 4MB (Dev board)

### Description:
Using Arduino IDE 1.8.13
Board selected : ESP32 Dev Module
Partition Scheme: "Default 4MB with spiffs(1.2MB APP/1.5MB SPIFFS)"
Sketch : ..\libraries\ESP32Encoder\examples\Encoder\Encoder.ino
Library from https://github.com/madhephaestus/ESP32Encoder version=0.5.1
Compilation error with Arduino core for the ESP32 release v2.0.1RC1
Error :
..\libraries\ESP32Encoder\src\ESP32Encoder.cpp:40:25: error: 'PCNT' was not declared in this scope
  uint32_t intr_status = PCNT.int_st.val;
                          ^~~~

It's working fine with Arduino core for the ESP32 release v2.0.0

### Sketch: 
```
#include <ESP32Encoder.h>

ESP32Encoder encoder;
ESP32Encoder encoder2;

// timer and flag for example, not needed for encoders
unsigned long encoder2lastToggled;
bool encoder2Paused = false;

void setup(){
	
	Serial.begin(115200);
	// Enable the weak pull down resistors

	//ESP32Encoder::useInternalWeakPullResistors=DOWN;
	// Enable the weak pull up resistors
	ESP32Encoder::useInternalWeakPullResistors=UP;

	// use pin 19 and 18 for the first encoder
	encoder.attachHalfQuad(19, 18);
	// use pin 17 and 16 for the second encoder
	encoder2.attachHalfQuad(17, 16);
		
	// set starting count value after attaching
	encoder.setCount(37);

	// clear the encoder's raw count and set the tracked count to zero
	encoder2.clearCount();
	Serial.println("Encoder Start = " + String((int32_t)encoder.getCount()));
	// set the lastToggle
	encoder2lastToggled = millis();
}

void loop(){
	// Loop and read the count
	Serial.println("Encoder count = " + String((int32_t)encoder.getCount()) + " " + String((int32_t)encoder2.getCount()));
	delay(100);

	// every 5 seconds toggle encoder 2
	if (millis() - encoder2lastToggled >= 5000) {
		if(encoder2Paused) {
			Serial.println("Resuming Encoder 2");
			encoder2.resumeCount();
		} else {
			Serial.println("Paused Encoder 2");
			encoder2.pauseCount();
		}

		encoder2Paused = !encoder2Paused;
		encoder2lastToggled = millis();
	}
}

```

### Debug Messages:
```
E:\Documents\me\Labo\Arduino\Croquis\libraries\ESP32Encoder\src\ESP32Encoder.cpp: In function 'void pcnt_example_intr_handler(void*)':
E:\Documents\me\Labo\Arduino\Croquis\libraries\ESP32Encoder\src\ESP32Encoder.cpp:40:25: error: 'PCNT' was not declared in this scope
  uint32_t intr_status = PCNT.int_st.val;
                         ^~~~
E:\Documents\me\Labo\Arduino\Croquis\libraries\ESP32Encoder\src\ESP32Encoder.cpp:40:25: note: suggested alternative: 'OCT'
  uint32_t intr_status = PCNT.int_st.val;
                         ^~~~
                         OCT
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

```

atanisoft, 2021-10-31:
This looks like a bug in the ESPEncoder library not including correct headers.

whogarden, 2021-11-01:
It's working fine with Arduino core for the ESP32 release v2.0.0
I am not an expert.
Can you tell me more about what can be uncorrect with ESP32 release v2.0.1RC1.
Thank you.

whogarden, 2021-11-01:
Please excuse me, 
I will send a warning to  madhephaestus/ESP32Encoder asking them to have a look to Pulse Counter...
Thank you

whogarden, 2021-11-10:
There is now a new version  https://github.com/madhephaestus/ESP32Encoder : 
version=0.6.0
Tested and working good.
I hope this can help other people

closed 2021-11-10 by whogarden