espressif/arduino-esp32 #5825: Arduino IDE throws error "arduino-builder returned 2" when trying to compile any sketch twice (even without any board/hardware connected) on any ESP32 board

furquan-lp, 2021-11-01:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.0 and 2.0.1-RC1
IDE name:						Arduino IDE 1.8.16
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Debian Linux

### Description:
The Arduino IDE throws an error (see below for detail) "arduino-builder returned 2 Error compiling for board [board name]." whenever **any** sketch is compiled twice for _any_ board selected from the Arduino-ESP32 core.

**No board is connected to the machine**, this appears to be a software problem.

This happens for any sketch as long as it contains a `#include`. Selecting a different board fixes this problem momentarily (the IDE will say `Build options changed, rebuilding all`) but compiling one more time will throw this error.
Examples of errors with random boards that I tried out:
```
arduino-builder returned 2
Error compiling for board ESP32C3 Dev Module.
```
```
arduino-builder returned 2
Error compiling for board ESP32 Wrover Kit (all versions).
```

Installed 2.0.0 and 2.0.1-RC1, problem occurs on **both versions**. Problem only occurs on Arduin-ESP32 core boards, Arduino and other boards work fine.

### Sketch: 
```cpp
#include <WiFi.h>

void setup() {
}

void loop() {
}
```

### Another Sketch (code must contain a `#include`):
```cpp
#include <Wire.h>
void setup() {}
void loop() {}
```

### Debug Messages:
```
Arduino: 1.8.16 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
panic: runtime error: index out of range [3] with length 3

goroutine 1 [running]:
arduino.cc/builder.(*includeCache).Next(...)
	arduino.cc/builder/container_find_includes.go:209
arduino.cc/builder.findIncludesUntilDone(0xc000038c00, 0xc00039e600, {{0x52ee20, 0xc00040f860}, {0xc00001622d, 0x7}})
	arduino.cc/builder/container_find_includes.go:321 +0x119d
arduino.cc/builder.(*ContainerFindIncludes).Run(0xc0000241c0, 0xc000038c00)
	arduino.cc/builder/container_find_includes.go:149 +0xc25
arduino.cc/builder.runCommands(0x8, {0xc000093c58, 0x22, 0x5555555}, 0x10)
	arduino.cc/builder/builder.go:191 +0xda
arduino.cc/builder.(*Builder).Run(0x7ffea35de365, 0x19)
	arduino.cc/builder/builder.go:124 +0x890
arduino.cc/builder.RunBuilder(...)
	arduino.cc/builder/builder.go:222
main.main()
	arduino.cc/arduino-builder/main.go:338 +0x988
arduino-builder returned 2
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

me-no-dev, 2021-11-01:
this seems to be an issue with your Arduino IDE or environment on your computer.

closed 2021-11-01 by me-no-dev

furquan-lp, 2021-11-01:
> this seems to be an issue with your Arduino IDE or environment on your computer.

Well, removing all the IDE files and reinstalling everything doesn't help either. Online there are some inconclusive reports of similar error messages ("panic: runtime error: index out of range [3] with length 3") but without any solution.

Moreover this only occurs with Arduino-ESP32 core boards. Stock Arduino boards and some other 3rd party boards compile fine.

lbernstone, 2021-11-01:
Removing all IDE files includes your .arduino15 directory?

furquan-lp, 2021-11-01:
> Removing all IDE files includes your .arduino15 directory?

Yes I just checked my command history and I ran `rm -rf .arduino15/` and `rm -rf Arduino` in home before reinstalling the IDE (and the ESP32 core).

me-no-dev, 2021-11-01:
can you try with older version? related to esp32 or not, the log does not give much clues, other than code requested an item from array with invalid index. As the log suggests, you can/should also enable compile debug in the IDE preferences.

furquan-lp, 2021-11-02:
> can you try with older version? related to esp32 or not, the log does not give much clues, other than code requested an item from array with invalid index. As the log suggests, you can/should also enable compile debug in the IDE preferences.

You were right. It was an environment issue. It turns out that the ESP32 core wasn't downloading correctly and was throwing a CRC error (similar to esp8266/Arduino#630), continuing the installation in such a case may have been causing a faulty install of the core. I'm betting on this being the cause because earlier when I had reintalled everything, the CRC error had come up again (and I ignored it again and continued with the install).

I fixed the issue by doing a fresh install of the Arduino IDE (completely removing `~/.arduino15` and `~/Arduino` as I had done before) and then installing the core using a _different connection_. The CRC error didn't come up this time and now the Arduino-ESP32 core works fine.