espressif/arduino-esp32 #5826: Memory leak on multiple calls to loadCACert and others on WiFiClientSecure

sguarin, 2021-11-01:
If you have a system that can accept certs configuration, then multiple calls to loadCACert or related functions will not reutilize the memory assigned.

https://github.com/sguarin/arduino-esp32/blob/caa8d07aafa04441bb85e6046a795249c01d9e39/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L304

_streamLoad()
allocates memory for new certificates. A better approach should be using realloc().

https://github.com/sguarin/arduino-esp32/blob/caa8d07aafa04441bb85e6046a795249c01d9e39/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L290

sguarin, 2021-11-27:
This temporary fix work for me.
But I think the memory pointer should not be static and should live on the class.

https://github.com/espressif/arduino-esp32/compare/master...sguarin:fix_streamLoad?expand=1

TD-er, 2021-11-27:
Just a repeat of the comment I placed with that commit:

You're not checking the return value of `realloc`.
If the return value is not a nullptr, you should assign it to `*destPtr`.
If it is a nullptr, it depends on the C++ version whether it was a failed alloc or not changed (e.g. size was the same)

And no need to check if `*destPtr` is a nullptr, as `realloc` called with a nullptr acts like malloc.

sguarin, 2021-12-25:
You are right. Thanks!

closed 2022-04-26 by me-no-dev