espressif/arduino-esp32 #5827: Build Failed for ESP32

pavithvishnu, 2021-11-02:
Hi,
I am using ESPidf(4.3.1) with  Arduino-esp32 lib(latest)

I'm getting errors, which are shown below

FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c.c.obj
ccache C:\Users\pavit\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino/variants/esp32 -I../components/arduino/cores/esp32 -I../components/arduino/libraries/ArduinoOTA/src -I../components/arduino/libraries/AsyncUDP/src -I../components/arduino/libraries/BLE/src -I../components/arduino/libraries/BluetoothSerial/src -I../components/arduino/libraries/DNSServer/src -I../components/arduino/libraries/EEPROM/src -I../components/arduino/libraries/ESP32/src -I../components/arduino/libraries/ESPmDNS/src -I../components/arduino/libraries/FFat/src -I../components/arduino/libraries/FS/src -I../components/arduino/libraries/HTTPClient/src -I../components/arduino/libraries/HTTPUpdate/src -I../components/arduino/libraries/LittleFS/src -I../components/arduino/libraries/NetBIOS/src -I../components/arduino/libraries/Preferences/src -I../components/arduino/libraries/RainMaker/src -I../components/arduino/libraries/SD_MMC/src -I../components/arduino/libraries/SD/src -I../components/arduino/libraries/SimpleBLE/src -I../components/arduino/libraries/SPIFFS/src -I../components/arduino/libraries/SPI/src -I../components/arduino/libraries/Ticker/src -I../components/arduino/libraries/Update/src -I../components/arduino/libraries/USB/src -I../components/arduino/libraries/WebServer/src -I../components/arduino/libraries/WiFiClientSecure/src -I../components/arduino/libraries/WiFi/src -I../components/arduino/libraries/WiFiProv/src -I../components/arduino/libraries/Wire/src -I../components/arduino/cores/esp32/libb64 -IG:/office/esp32/components/newlib/platform_include -IG:/office/esp32/components/freertos/include -IG:/office/esp32/components/freertos/port/xtensa/include -IG:/office/esp32/components/esp_hw_support/include -IG:/office/esp32/components/esp_hw_support/port/esp32/. -IG:/office/esp32/components/heap/include -IG:/office/esp32/components/log/include -IG:/office/esp32/components/lwip/include/apps -IG:/office/esp32/components/lwip/include/apps/sntp -IG:/office/esp32/components/lwip/lwip/src/include -IG:/office/esp32/components/lwip/port/esp32/include -IG:/office/esp32/components/lwip/port/esp32/include/arch -IG:/office/esp32/components/soc/include -IG:/office/esp32/components/soc/esp32/. -IG:/office/esp32/components/soc/esp32/include -IG:/office/esp32/components/hal/esp32/include -IG:/office/esp32/components/hal/include -IG:/office/esp32/components/esp_rom/include -IG:/office/esp32/components/esp_rom/esp32 -IG:/office/esp32/components/esp_rom/include/esp32 -IG:/office/esp32/components/esp_common/include -IG:/office/esp32/components/esp_system/include -IG:/office/esp32/components/esp32/include -IG:/office/esp32/components/driver/include -IG:/office/esp32/components/driver/esp32/include -IG:/office/esp32/components/esp_ringbuf/include -IG:/office/esp32/components/efuse/include -IG:/office/esp32/components/efuse/esp32/include -IG:/office/esp32/components/xtensa/include -IG:/office/esp32/components/xtensa/esp32/include -IG:/office/esp32/components/espcoredump/include -IG:/office/esp32/components/esp_timer/include -IG:/office/esp32/components/esp_ipc/include -IG:/office/esp32/components/esp_pm/include -IG:/office/esp32/components/vfs/include -IG:/office/esp32/components/esp_wifi/include -IG:/office/esp32/components/esp_wifi/esp32/include -IG:/office/esp32/components/esp_event/include -IG:/office/esp32/components/esp_netif/include -IG:/office/esp32/components/esp_eth/include -IG:/office/esp32/components/tcpip_adapter/include -IG:/office/esp32/components/app_trace/include -IG:/office/esp32/components/spi_flash/include -IG:/office/esp32/components/mbedtls/port/include -IG:/office/esp32/components/mbedtls/mbedtls/include -IG:/office/esp32/components/mbedtls/esp_crt_bundle/include -IG:/office/esp32/components/mdns/include -IG:/office/esp32/components/console -IG:/office/esp32/components/esp_adc_cal/include -IG:/office/esp32/components/wifi_provisioning/include -IG:/office/esp32/components/protocomm/include/common -IG:/office/esp32/components/protocomm/include/security -IG:/office/esp32/components/protocomm/include/transports -IG:/office/esp32/components/nvs_flash/include -IG:/office/esp32/components/nghttp/port/include -IG:/office/esp32/components/nghttp/nghttp2/lib/includes -IG:/office/esp32/components/fatfs/diskio -IG:/office/esp32/components/fatfs/vfs -IG:/office/esp32/components/fatfs/src -IG:/office/esp32/components/wear_levelling/include -IG:/office/esp32/components/sdmmc/include -IG:/office/esp32/components/app_update/include -IG:/office/esp32/components/bootloader_support/include -IG:/office/esp32/components/spiffs/include -IG:/office/esp32/components/openssl/include -IG:/office/esp32/components/esp_hid/include -IG:/office/esp32/components/esp_https_ota/include -IG:/office/esp32/components/esp_http_client/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.3.1\" -DESP_PLATFORM -DARDUINO=10812 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32_DEV\" -DARDUINO_VARIANT=\"esp32\" -DESP32 -MD -MT esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c.c.obj -MF esp-idf\arduino\CMakeFiles\__idf_arduino.dir\cores\esp32\esp32-hal-i2c.c.obj.d -o esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c.c.obj   -c ../components/arduino/cores/esp32/esp32-hal-i2c.c
../components/arduino/cores/esp32/esp32-hal-i2c.c: In function 'i2cWrite':
../components/arduino/cores/esp32/esp32-hal-i2c.c:152:22: error: implicit declaration of function 'I2C_LINK_RECOMMENDED_SIZE' [-Werror=implicit-function-declaration]
     uint8_t cmd_buff[I2C_LINK_RECOMMENDED_SIZE(1)] = { 0 };
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/esp32-hal-i2c.c:152:5: error: variable-sized object may not be initialized
     uint8_t cmd_buff[I2C_LINK_RECOMMENDED_SIZE(1)] = { 0 };
     ^~~~~~~
../components/arduino/cores/esp32/esp32-hal-i2c.c:152:56: warning: excess elements in array initializer
     uint8_t cmd_buff[I2C_LINK_RECOMMENDED_SIZE(1)] = { 0 };
                                                        ^
../components/arduino/cores/esp32/esp32-hal-i2c.c:152:56: note: (near initialization for 'cmd_buff')
../components/arduino/cores/esp32/esp32-hal-i2c.c:153:11: error: implicit declaration of function 'i2c_cmd_link_create_static'; did you mean 'i2c_cmd_link_create'? [-Werror=implicit-function-declaration]
     cmd = i2c_cmd_link_create_static(cmd_buff, I2C_LINK_RECOMMENDED_SIZE(1));
           ^~~~~~~~~~~~~~~~~~~~~~~~~~
           i2c_cmd_link_create
../components/arduino/cores/esp32/esp32-hal-i2c.c:153:9: warning: assignment to 'i2c_cmd_handle_t' {aka 'void *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
     cmd = i2c_cmd_link_create_static(cmd_buff, I2C_LINK_RECOMMENDED_SIZE(1));
         ^
../components/arduino/cores/esp32/esp32-hal-i2c.c:145:9: error: jump into scope of identifier with variably modified type
         goto end;
         ^~~~
../components/arduino/cores/esp32/esp32-hal-i2c.c:174:1: note: label 'end' defined here
 end:
 ^~~
../components/arduino/cores/esp32/esp32-hal-i2c.c:152:13: note: 'cmd_buff' declared here
     uint8_t cmd_buff[I2C_LINK_RECOMMENDED_SIZE(1)] = { 0 };
             ^~~~~~~~
../components/arduino/cores/esp32/esp32-hal-i2c.c:176:9: error: implicit declaration of function 'i2c_cmd_link_delete_static'; did you mean 'i2c_cmd_link_delete'? [-Werror=implicit-function-declaration]
         i2c_cmd_link_delete_static(cmd);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
         i2c_cmd_link_delete
../components/arduino/cores/esp32/esp32-hal-i2c.c: In function 'i2cRead':
../components/arduino/cores/esp32/esp32-hal-i2c.c:200:15: error: implicit declaration of function 'i2c_master_read_from_device'; did you mean 'i2c_master_read_byte'? [-Werror=implicit-function-declaration]
[... cut ...]
[966/1098] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c-slave.c.obj
../components/arduino/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2c_slave_isr_handler':
../components/arduino/cores/esp32/esp32-hal-i2c-slave.c:694:13: warning: unused variable 'tx_fifo_len' [-Wunused-variable]
     uint8_t tx_fifo_len = SOC_I2C_FIFO_LEN - i2c_ll_get_txfifo_len(i2c->dev);
             ^~~~~~~~~~~
[968/1098] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-bt.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

me-no-dev, 2021-11-02:
Use ESP-IDF v4.4

closed 2021-11-02 by me-no-dev

VojtechBartoska, 2021-11-03:
Duplicate of #5819