espressif/arduino-esp32 #5832: Interpretation of Exeption Decoder Output

Paulsburg, 2021-11-02:
### Hardware:
Board: ESP32 Dev Module TTGO SIM800, produced by LILYGO
          Chip is ESP32-D0WD (revision 1)
          Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
          Crystal is 40MHz
IDE: Arduino IDE 1.8.16

### Description:
I can not interpret the output of the Exception Decoder Output.

Perhaps somebody can help...

Thanks
Axel

The Guru Meditation Error looks likes this:
--------------------------------------------------------------------------------------------

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x400eae82  PS      : 0x00060936  A0      : 0x800eafeb  A1      : 0x3ffc5960  
A2      : 0x3ffc5b04  A3      : 0x00000016  A4      : 0x3ffc495c  A5      : 0x0f6e5898  
A6      : 0x00000000  A7      : 0xff000000  A8      : 0x25a5a5a5  A9      : 0x3ffc5c70  
A10     : 0x00000000  A11     : 0x3ffc5d54  A12     : 0x3ffde2f8  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  

ELF file SHA256: 0000000000000000

Backtrace: 0x400eae82:0x3ffc5960 0x400eafe8:0x3ffc5990 0x400eaffd:0x3ffc59b0 0x400eb043:0x3ffc59d0 0x400d4805:0x3ffc59f0 0x400d512e:0x3ffc5d40 0x400d7f86:0x3ffc5d90 0x400e30ea:0x3ffc5f50 0x400e8d12:0x3ffc7680 0x400eb7c2:0x3ffc78b0 0x40090962:0x3ffc78d0

The output of the Exception Decoder looks like this:
---------------------------------------------------------------------------------------------

PC: 0x400eae82: String::changeBuffer(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 155
EXCVADDR: 0x00000000

Decoding stack results
0x400eae82: String::changeBuffer(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 155
0x400eafe8: String::reserve(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 149
0x400eaffd: String::reserve(unsigned int) at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 153
0x400eb043: String::String(char const*) at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 35
0x400d4805: waitResponse_search(String, unsigned int, bool) at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/WString.h line 341
0x400d512e: sendATCommand_String(String, String, int, bool) at C:\Users\...\Documents\Arduino\ESP32_deepsleep_Fallenmelder_12b6/SIM800_ctrl.h line 561
0x400d7f86: Send_SMS(String, unsigned long long, bool) at C:\Users\...\Documents\Arduino\ESP32_deepsleep_Fallenmelder_12b6/SIM800_SMS.h line 34
0x400e30ea: reaktion_OnWakeUpReason(unsigned char, int, String, String, String) at C:\Users\...\Documents\Arduino\ESP32_deepsleep_Fallenmelder_12b6/SwitchOnWakeUpReason.h line 573
0x400e8d12: setup() at C:\Users\...\Documents\Arduino\ESP32_deepsleep_Fallenmelder_12b6/ESP32_deepsleep_Fallenmelder_12b6.ino line 505
0x400eb7c2: _GLOBAL__sub_D__ZN6StringC2EPKc() at C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 124
0x40090962: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

lbernstone, 2021-11-02:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html
You have overflowed your stack (typically 8K).  Stop passing Strings between functions, that's what pointers and references are for.

Paulsburg, 2021-11-03:
Thanks for the info

Its a lot of code, everywhere using Strings...

I will try to rewrite / change to pionters

the Espressiv Doc is topp

Thanks

Paulsburg, 2021-11-09:
Problem solved by using  a lot of calls_by_reference in most of my functions.
Thanks for the help

closed 2021-11-09 by Paulsburg