espressif/arduino-esp32 #5833: Arduino as a ESP-IDF component Build Failure

TrikkStar, 2021-11-02:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Complete 
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32-WROOVER-IE
Core Installation version:			ESP-IDF v5.0-dev-79-gbcbef9a8db
IDE name:						ESP-IDF CLI
Computer OS:                                         Debian Bullseye running under WSL2

### Description:
Trying to complete the [getting started instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#installation).

### Sketch: 
```cpp

#include "Arduino.h"

void setup(){
    Serial.begin(115200);
}

void loop(){
    Serial.println("loop");
    delay(1000);
}
```

### Debug Messages:
```
[1019/1229] Building C object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-timer.c.obj
FAILED: esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-timer.c.obj
/home/nthederahn/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino-esp32/variants/esp32 -I../components/arduino-esp32/cores/esp32 -I../components/arduino-esp32/libraries/ArduinoOTA/src -I../components/arduino-esp32/libraries/AsyncUDP/src -I../components/arduino-esp32/libraries/BLE/src -I../components/arduino-esp32/libraries/BluetoothSerial/src -I../components/arduino-esp32/libraries/DNSServer/src -I../components/arduino-esp32/libraries/EEPROM/src -I../components/arduino-esp32/libraries/ESP32/src -I../components/arduino-esp32/libraries/ESPmDNS/src -I../components/arduino-esp32/libraries/FFat/src -I../components/arduino-esp32/libraries/FS/src -I../components/arduino-esp32/libraries/HTTPClient/src -I../components/arduino-esp32/libraries/HTTPUpdate/src -I../components/arduino-esp32/libraries/LittleFS/src -I../components/arduino-esp32/libraries/NetBIOS/src -I../components/arduino-esp32/libraries/Preferences/src -I../components/arduino-esp32/libraries/RainMaker/src -I../components/arduino-esp32/libraries/SD_MMC/src -I../components/arduino-esp32/libraries/SD/src -I../components/arduino-esp32/libraries/SimpleBLE/src -I../components/arduino-esp32/libraries/SPIFFS/src -I../components/arduino-esp32/libraries/SPI/src -I../components/arduino-esp32/libraries/Ticker/src -I../components/arduino-esp32/libraries/Update/src -I../components/arduino-esp32/libraries/USB/src -I../components/arduino-esp32/libraries/WebServer/src -I../components/arduino-esp32/libraries/WiFiClientSecure/src -I../components/arduino-esp32/libraries/WiFi/src -I../components/arduino-esp32/libraries/WiFiProv/src -I../components/arduino-esp32/libraries/Wire/src -I../components/arduino-esp32/cores/esp32/libb64 -I/home/nthederahn/esp/esp-idf/components/newlib/platform_include -I/home/nthederahn/esp/esp-idf/components/freertos/include -I/home/nthederahn/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/nthederahn/esp/esp-idf/components/freertos/port/xtensa/include -I/home/nthederahn/esp/esp-idf/components/freertos/include/esp_additions -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/include -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/include/soc -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/nthederahn/esp/esp-idf/components/heap/include -I/home/nthederahn/esp/esp-idf/components/log/include -I/home/nthederahn/esp/esp-idf/components/lwip/include/apps -I/home/nthederahn/esp/esp-idf/components/lwip/include/apps/sntp -I/home/nthederahn/esp/esp-idf/components/lwip/lwip/src/include -I/home/nthederahn/esp/esp-idf/components/lwip/port/esp32/include -I/home/nthederahn/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/nthederahn/esp/esp-idf/components/soc/include -I/home/nthederahn/esp/esp-idf/components/soc/esp32/. -I/home/nthederahn/esp/esp-idf/components/soc/esp32/include -I/home/nthederahn/esp/esp-idf/components/hal/esp32/include -I/home/nthederahn/esp/esp-idf/components/hal/include -I/home/nthederahn/esp/esp-idf/components/hal/platform_port/include -I/home/nthederahn/esp/esp-idf/components/esp_rom/include -I/home/nthederahn/esp/esp-idf/components/esp_rom/include/esp32 -I/home/nthederahn/esp/esp-idf/components/esp_rom/esp32 -I/home/nthederahn/esp/esp-idf/components/esp_common/include -I/home/nthederahn/esp/esp-idf/components/esp_system/include -I/home/nthederahn/esp/esp-idf/components/esp_system/port/soc -I/home/nthederahn/esp/esp-idf/components/esp_system/port/public_compat -I/home/nthederahn/esp/esp-idf/components/esp32/include -I/home/nthederahn/esp/esp-idf/components/xtensa/include -I/home/nthederahn/esp/esp-idf/components/xtensa/esp32/include -I/home/nthederahn/esp/esp-idf/components/driver/include -I/home/nthederahn/esp/esp-idf/components/driver/esp32/include -I/home/nthederahn/esp/esp-idf/components/esp_pm/include -I/home/nthederahn/esp/esp-idf/components/esp_ringbuf/include -I/home/nthederahn/esp/esp-idf/components/efuse/include -I/home/nthederahn/esp/esp-idf/components/efuse/esp32/include -I/home/nthederahn/esp/esp-idf/components/vfs/include -I/home/nthederahn/esp/esp-idf/components/esp_wifi/include -I/home/nthederahn/esp/esp-idf/components/esp_event/include -I/home/nthederahn/esp/esp-idf/components/esp_netif/include -I/home/nthederahn/esp/esp-idf/components/esp_eth/include -I/home/nthederahn/esp/esp-idf/components/tcpip_adapter/include -I/home/nthederahn/esp/esp-idf/components/esp_phy/include -I/home/nthederahn/esp/esp-idf/components/esp_phy/esp32/include -I/home/nthederahn/esp/esp-idf/components/esp_ipc/include -I/home/nthederahn/esp/esp-idf/components/app_trace/include -I/home/nthederahn/esp/esp-idf/components/esp_timer/include -I/home/nthederahn/esp/esp-idf/components/spi_flash/include -I/home/nthederahn/esp/esp-idf/components/mbedtls/port/include -I/home/nthederahn/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/nthederahn/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/nthederahn/esp/esp-idf/components/mdns/include -I/home/nthederahn/esp/esp-idf/components/console -I/home/nthederahn/esp/esp-idf/components/esp_adc_cal/include -I/home/nthederahn/esp/esp-idf/components/wifi_provisioning/include -I/home/nthederahn/esp/esp-idf/components/protocomm/include/common -I/home/nthederahn/esp/esp-idf/components/protocomm/include/security -I/home/nthederahn/esp/esp-idf/components/protocomm/include/transports -I/home/nthederahn/esp/esp-idf/components/nvs_flash/include -I/home/nthederahn/esp/esp-idf/components/nghttp/port/include -I/home/nthederahn/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/nthederahn/esp/esp-idf/components/fatfs/diskio -I/home/nthederahn/esp/esp-idf/components/fatfs/vfs -I/home/nthederahn/esp/esp-idf/components/fatfs/src -I/home/nthederahn/esp/esp-idf/components/wear_levelling/include -I/home/nthederahn/esp/esp-idf/components/sdmmc/include -I/home/nthederahn/esp/esp-idf/components/app_update/include -I/home/nthederahn/esp/esp-idf/components/bootloader_support/include -I/home/nthederahn/esp/esp-idf/components/spiffs/include -I/home/nthederahn/esp/esp-idf/components/openssl/include -I/home/nthederahn/esp/esp-idf/components/esp_hid/include -I/home/nthederahn/esp/esp-idf/components/esp_https_ota/include -I/home/nthederahn/esp/esp-idf/components/esp_http_client/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/mnt/c/Users/nthederahn/Projects/moscow-initializer/init-firmware=. -fmacro-prefix-map=/home/nthederahn/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-de [... cut ...]
[1023/1229] Building C object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-rmt.c.obj
../components/arduino-esp32/cores/esp32/esp32-hal-rmt.c:118:25: warning: 'g_rmt_objlocks' defined but not used [-Wunused-variable]
 static xSemaphoreHandle g_rmt_objlocks[MAX_CHANNELS] = {
                         ^~~~~~~~~~~~~~
[1024/1229] Building CXX object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/MD5Builder.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

chegewara, 2021-11-02:
Not exactly related, but explaining issue:
https://github.com/espressif/esp-iot-solution/issues/133

atanisoft, 2021-11-02:
arduino-esp32 master uses the esp-idf release/v4.4 branch at this time as well.

TrikkStar, 2021-11-02:
> arduino-esp32 master uses the esp-idf release/v4.4 branch at this time as well.

Yeah, I had tried with 4.4 and got a different error
```
[1007/1229] Building C object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-i2c-slave.c.obj
FAILED: esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-i2c-slave.c.obj
/home/nthederahn/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino-esp32/variants/esp32 -I../components/arduino-esp32/cores/esp32 -I../components/arduino-esp32/libraries/ArduinoOTA/src -I../components/arduino-esp32/libraries/AsyncUDP/src -I../components/arduino-esp32/libraries/BLE/src -I../components/arduino-esp32/libraries/BluetoothSerial/src -I../components/arduino-esp32/libraries/DNSServer/src -I../components/arduino-esp32/libraries/EEPROM/src -I../components/arduino-esp32/libraries/ESP32/src -I../components/arduino-esp32/libraries/ESPmDNS/src -I../components/arduino-esp32/libraries/FFat/src -I../components/arduino-esp32/libraries/FS/src -I../components/arduino-esp32/libraries/HTTPClient/src -I../components/arduino-esp32/libraries/HTTPUpdate/src -I../components/arduino-esp32/libraries/LittleFS/src -I../components/arduino-esp32/libraries/NetBIOS/src -I../components/arduino-esp32/libraries/Preferences/src -I../components/arduino-esp32/libraries/RainMaker/src -I../components/arduino-esp32/libraries/SD_MMC/src -I../components/arduino-esp32/libraries/SD/src -I../components/arduino-esp32/libraries/SimpleBLE/src -I../components/arduino-esp32/libraries/SPIFFS/src -I../components/arduino-esp32/libraries/SPI/src -I../components/arduino-esp32/libraries/Ticker/src -I../components/arduino-esp32/libraries/Update/src -I../components/arduino-esp32/libraries/USB/src -I../components/arduino-esp32/libraries/WebServer/src -I../components/arduino-esp32/libraries/WiFiClientSecure/src -I../components/arduino-esp32/libraries/WiFi/src -I../components/arduino-esp32/libraries/WiFiProv/src -I../components/arduino-esp32/libraries/Wire/src -I../components/arduino-esp32/cores/esp32/libb64 -I/home/nthederahn/esp/esp-idf/components/newlib/platform_include -I/home/nthederahn/esp/esp-idf/components/freertos/include -I/home/nthederahn/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/nthederahn/esp/esp-idf/components/freertos/port/xtensa/include -I/home/nthederahn/esp/esp-idf/components/freertos/include/esp_additions -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/include -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/include/soc -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/nthederahn/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/nthederahn/esp/esp-idf/components/heap/include -I/home/nthederahn/esp/esp-idf/components/log/include -I/home/nthederahn/esp/esp-idf/components/lwip/include/apps -I/home/nthederahn/esp/esp-idf/components/lwip/include/apps/sntp -I/home/nthederahn/esp/esp-idf/components/lwip/lwip/src/include -I/home/nthederahn/esp/esp-idf/components/lwip/port/esp32/include -I/home/nthederahn/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/nthederahn/esp/esp-idf/components/soc/include -I/home/nthederahn/esp/esp-idf/components/soc/esp32/. -I/home/nthederahn/esp/esp-idf/components/soc/esp32/include -I/home/nthederahn/esp/esp-idf/components/hal/esp32/include -I/home/nthederahn/esp/esp-idf/components/hal/include -I/home/nthederahn/esp/esp-idf/components/hal/platform_port/include -I/home/nthederahn/esp/esp-idf/components/esp_rom/include -I/home/nthederahn/esp/esp-idf/components/esp_rom/include/esp32 -I/home/nthederahn/esp/esp-idf/components/esp_rom/esp32 -I/home/nthederahn/esp/esp-idf/components/esp_common/include -I/home/nthederahn/esp/esp-idf/components/esp_system/include -I/home/nthederahn/esp/esp-idf/components/esp_system/port/soc -I/home/nthederahn/esp/esp-idf/components/esp_system/port/public_compat -I/home/nthederahn/esp/esp-idf/components/esp32/include -I/home/nthederahn/esp/esp-idf/components/xtensa/include -I/home/nthederahn/esp/esp-idf/components/xtensa/esp32/include -I/home/nthederahn/esp/esp-idf/components/driver/include -I/home/nthederahn/esp/esp-idf/components/driver/esp32/include -I/home/nthederahn/esp/esp-idf/components/esp_pm/include -I/home/nthederahn/esp/esp-idf/components/esp_ringbuf/include -I/home/nthederahn/esp/esp-idf/components/efuse/include -I/home/nthederahn/esp/esp-idf/components/efuse/esp32/include -I/home/nthederahn/esp/esp-idf/components/vfs/include -I/home/nthederahn/esp/esp-idf/components/esp_wifi/include -I/home/nthederahn/esp/esp-idf/components/esp_event/include -I/home/nthederahn/esp/esp-idf/components/esp_netif/include -I/home/nthederahn/esp/esp-idf/components/esp_eth/include -I/home/nthederahn/esp/esp-idf/components/tcpip_adapter/include -I/home/nthederahn/esp/esp-idf/components/esp_phy/include -I/home/nthederahn/esp/esp-idf/components/esp_phy/esp32/include -I/home/nthederahn/esp/esp-idf/components/esp_ipc/include -I/home/nthederahn/esp/esp-idf/components/app_trace/include -I/home/nthederahn/esp/esp-idf/components/esp_timer/include -I/home/nthederahn/esp/esp-idf/components/spi_flash/include -I/home/nthederahn/esp/esp-idf/components/mbedtls/port/include -I/home/nthederahn/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/nthederahn/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/h [... cut ...]
[1022/1229] Building C object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/esp32-hal-rmt.c.obj
../components/arduino-esp32/cores/esp32/esp32-hal-rmt.c:118:25: warning: 'g_rmt_objlocks' defined but not used [-Wunused-variable]
 static xSemaphoreHandle g_rmt_objlocks[MAX_CHANNELS] = {
                         ^~~~~~~~~~~~~~
[1024/1229] Building CXX object esp-idf/arduino-esp32/CMakeFiles/__idf_arduino-esp32.dir/cores/esp32/MD5Builder.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

atanisoft, 2021-11-02:
Check your setting for `CONFIG_DISABLE_HAL_LOCKS`, it looks like there are a few unprotected references (@me-no-dev FYI)

TrikkStar, 2021-11-02:
`CONFIG_DISABLE_HAL_LOCKS=y`

atanisoft, 2021-11-02:
Remove that setting and it should work.

TrikkStar, 2021-11-03:
It does compile now.  Think this part of the documentation might need adjusted, as to me it implied that I should have checked that option:

> “Disable mutex locks for HAL”
> 
> If enabled, there will be no protection on the drivers from concurently accessing them from another thread/interrupt/core

VojtechBartoska, 2021-11-03:
Duplicate of #5819

VojtechBartoska, 2021-11-03:
Just missed the last [comment](https://github.com/espressif/arduino-esp32/issues/5833#issuecomment-959272241). Reopening, PTAL @me-no-dev, if it's relevant?

me-no-dev, 2021-11-04:
@VojtechBartoska yeah. seems like a bug :)

mrengineer7777, 2022-01-25:
The "lock" issue is resolved by https://github.com/espressif/arduino-esp32/pull/6108.  Unfortunately I also had lots of other compile issues (conflicting defines) using Arduino as IDF component in VS with Platformio.  I changed my platform to IDF and manually copied in and fixed the Arduino libraries that I wanted.

PilnyTomas, 2022-01-27:
I'm confused - is it solved, is there still a problem that needs to be solved? What is the current problem and how can I replicate it?

VojtechBartoska, 2022-01-27:
it was about the last [comment](https://github.com/espressif/arduino-esp32/issues/5833#issuecomment-959272241) from @TrikkStar.

Up to the @mrengineer7777 update, we need to make sure that #6108 really fix the problem with "lock".

mrengineer7777, 2022-01-27:
@VojtechBartoska @TrikkStar 
“Disable mutex locks for HAL”  - Yes that is in the espressif-arduino documentation somewhere.  Seems like bad advice, but some of the Arduino libraries are coded for that, and I believe that is how the espressif-arduino binaries are normally built.  Most are coded such that it will work with the CONFIG_DISABLE_HAL_LOCKS option on or off.  Pull #6108 fixes this specific compile problem and thus I believe closes this issue.  Should HAL locks be disabled is a much bigger discussion.

VojtechBartoska, 2022-01-27:
related note in Documentation about Hal locks is [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html?highlight=Disable%20mutex%20locks%20for%20HAL#option-2-using-esp-idf-appmain)

PR #6299 (2022-02-23): Update doc: Arduino as IDF component; fixes #5833
## Summary
Updated documentation describing the usage of Arduino-esp32 core as ESP-IDF component.

## Impact
Removed confusing mentions of advanced menuconfig options.
Extended process of installation, setup, and usage.

## Related links
Closes #5833
commit b5f3d6c8 (2022-02-23): Update doc: Arduino as IDF component; fixes #5833 (#6299)

Summary

Updated documentation describing the usage of Arduino-esp32 core as ESP-IDF component.

Impact

Removed confusing mentions of advanced menuconfig options.
Extended process of installation, setup, and usage.

Related links

Closes #5833