espressif/arduino-esp32 #5834: While outputting 5/10/20MHz master clock from LEDC or I2S to peripheral device, the Wifi throughput gets worse.

sohtamei, 2021-11-03:
### Hardware:
Board:							DevkitC/Espressif ESP32camera/ESP32-CAM(3rd party)/M5Camera/.. other devices
Core Installation version:			1.04
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					variable for target device
Computer OS:                                        Windows 10

### Description:
While outputting 5/10/20MHz master clock (continuous clock) from LEDC or I2S to peripheral device, the Wifi throughput gets worse.
While outputting 8MHz master clock, the Wifi throughput does not get worse.

**Please see following video (on facebook)**
https://www.facebook.com/100009187204902/videos/954929518394522/

### Question
I would like to know 
- Why 5/10/20MHz clock from ESP32 makes WiFi throughput worse,
- Is there any better solution for this issue than XCLK=8MHz.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
[Source code]
https://github.com/sohtamei/TuKuRutch.ext/tree/master/libraries/testCamWifi/src

**[binary, burning and configuring environment]**
1. burn firmware (customized CameraWebServer)
https://sohta02.sakura.ne.jp/tukurutch/
[Add extension]  
-- [M5Camera,ESP32cam,..]  
-- pulldown [burn (M5Camera)] to "testXclockIssue"  
-- click [burn (testXclockIssue)]

2. enter SSID and pass to [connect ssid ..] and click it.

3. click [Wifi Status] and get IP address
[image]

4. access to IP address of step.3
[image]

### Result
**XCLK=10MHz, test FW(above Sketch), without Camera(dummy network camera)**
[image]
The result of devices which are using chip-ESP32 is OK (no throughput decline).
The result of devices which are using WROOM/WROVER is NG, except for "aitendo board".

**XCLK=8MHz, customized CameraWebServer, with ov2640**
[image]
The result of all devices is OK.

sohtamei, 2021-11-09:
I fixed this issue with XCLK=8MHz and x2 clock on ov2640 side.

1. burn firmware (customized CameraWebServer)
https://sohta02.sakura.ne.jp/tukurutch/
[Add extension]
-- [M5Camera,ESP32cam,..]
-- pulldown [burn (M5Camera)] to "esp32cam"
-- click [burn (unitCam)]

2. enter SSID and pass to [connect ssid ..] and click it.

3. click [Wifi Status] and get IP address

4. access to IP address via web Browser

VojtechBartoska, 2022-04-07:
Hello @sohtamei, can you please test this with v2.0.3-RC1? TY

VojtechBartoska, 2022-05-04:
I'm closing this issue as expired due to no answer and mentioned Peripherals were refactored lately.

If it's still needed please reopen the issue.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska

xam-ps, 2023-01-07:
Setting the clock speed (xclk) to 8 Mhz did the job for me. Now getting constantly good framerates 👍

schmandforke, 2024-01-09:
Also fixed the issue for me in esphome:
```
esp32_camera:
  external_clock:
    pin: GPIO21
    frequency: 8MHz
```

ICMP perspective before:
```
64 bytes from 172.25.0.100: icmp_seq=480 ttl=255 time=3148.200 ms
64 bytes from 172.25.0.100: icmp_seq=481 ttl=255 time=2143.600 ms
64 bytes from 172.25.0.100: icmp_seq=482 ttl=255 time=1138.562 ms
64 bytes from 172.25.0.100: icmp_seq=483 ttl=255 time=802.870 ms
64 bytes from 172.25.0.100: icmp_seq=487 ttl=255 time=666.481 ms
64 bytes from 172.25.0.100: icmp_seq=488 ttl=255 time=272.646 ms
64 bytes from 172.25.0.100: icmp_seq=489 ttl=255 time=394.262 ms
```

and after:
```
64 bytes from 172.25.0.100: icmp_seq=843 ttl=255 time=10.845 ms
64 bytes from 172.25.0.100: icmp_seq=844 ttl=255 time=7.081 ms
64 bytes from 172.25.0.100: icmp_seq=845 ttl=255 time=8.416 ms
64 bytes from 172.25.0.100: icmp_seq=846 ttl=255 time=6.939 ms
64 bytes from 172.25.0.100: icmp_seq=847 ttl=255 time=6.320 ms
```

Great job guys 👍