espressif/arduino-esp32 #5837: How to read multiple ADC channels by using I2S in Arduino framework

IamMrUnderstand, 2021-11-03:
### Hardware:
|||
|:---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>Version/Date</B>|1.0.6|
|<B>IDE name</B>|Arduino IDE|
|<B>Flash Frequency</B>|80Mhz|
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|921600|
|<B>Computer OS</B>|Windows 10|

### Description:
I need an example read multiple ADC channels by using I2S in Arduino framework.
I have tried using "adc_i2s_pattern_t" but it's not in I2S.h

How can I read multiple ADC channels?

thanks.

### Sketch:
```cpp
#include <driver/i2s.h>

#define I2S_SAMPLE_RATE (6000)

uint16_t adc_reading;

void i2sInit() {
  i2s_config_t i2s_config = {
   .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
   .sample_rate =  I2S_SAMPLE_RATE,              // The format of the signal using ADC_BUILT_IN
   .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB
   .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
   .communication_format = I2S_COMM_FORMAT_I2S_MSB,
   .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
   .dma_buf_count = 4,
   .dma_buf_len = 8,
   .use_apll = false,
   .tx_desc_auto_clear = false,
   .fixed_mclk = 0
  };
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  //i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);
   
  static const adc_i2s_pattern_t adc_i2s_pattern[] = {
    {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_0},
    {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_3},
    {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_4},
    {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_5},
    {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_6},
    {.atten = ADC_ATTEN_DB_11, .bits = ADC_WIDTH_BIT_12, .channel = ADC1_CHANNEL_7}
  };
  
   i2s_adc_enable(I2S_NUM_0);
}

void reader(void *pvParameters) {
  size_t bytes_read;
  while(1){
    uint16_t buffer[6] = {0};
    i2s_read(I2S_NUM_0, &buffer, sizeof(buffer), &bytes_read, 15);
    if (bytes_read == sizeof(buffer)) {
        
    } else {
      Serial.println("buffer empty");
    }
  }
}

void setup() {
  Serial.begin(1000000);

  i2sInit();
  // Create a task that will read the data
  xTaskCreatePinnedToCore(reader, "ADC_reader", 2048, NULL, 1, NULL, 1);
}

void loop() {
}
```

### Debug Messages:
```
C:\Users\Top\Desktop\ESP32_ADC_TEST\ESP32_ADC_TEST.ino: In function 'void i2sInit()':
ESP32_ADC_TEST:29:16: error: 'adc_i2s_pattern_t' does not name a type
   static const adc_i2s_pattern_t adc_i2s_pattern[] = {
                ^
exit status 1
'adc_i2s_pattern_t' does not name a type
```

me-no-dev, 2021-11-04:
Have you tried this with 2.0.0?

IamMrUnderstand, 2021-11-04:
> Have you tried this with 2.0.0?
yes, I also have tired with 2.0.1-RC1

and got this
```
C:\Users\Top\Desktop\ESP32_ADC_TEST2\ESP32_ADC_TEST2.ino: In function 'void i2sInit()':
ESP32_ADC_TEST2:25:16: error: 'adc_i2s_pattern_t' does not name a type; did you mean 'adc_atten_t'?
   static const adc_i2s_pattern_t adc_i2s_pattern[] = {
                ^~~~~~~~~~~~~~~~~
                adc_atten_t
exit status 1
'adc_i2s_pattern_t' does not name a type; did you mean 'adc_atten_t'?
```

closed 2021-11-05 by IamMrUnderstand