espressif/arduino-esp32 #5838: AsyncUDP flaky results with IPv6, sometimes no packets are send

fapolumi, 2021-11-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			        2.0.0
IDE name:						Arduino IDE 1.8.16
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                          Linux

### Description:
I'm trying to reply to UDP packets with AsyncUDP. IPv4 is fine, IPv6 works only sometimes, though I can't spot, when and why. I guess it's a timing/setup related thing.

Details:
A python program on my linux box sends IP4 and IP6 packets (see below). Those are received perfectly fine by the ESP, both via the AP and STA interface depending on the connection used. The ESP sketch should reply to both of these packets, and does so for the IP4 packet 100% of the time. IP6 is the flaky part: Changing parts of WiFi related or unrelated code, recompilation or plain ESP resets with no code may result in IP6 packets not being send...

I used tcpdump as well to confirm the python output. No IP6 packets are to be seen for "bad builds/runs".  

### Sketch:
```cpp

#include "WiFi.h"
#include "AsyncUDP.h"

const char* ssid_ap = "*";
const char* pass_ap = "*";
const char* ssid = "*";
const char* password = "*";

bool canstart = false;
bool services = false;
bool shallstart = false;

AsyncUDP udp;

void WiFiEvent(WiFiEvent_t event) {  
  switch (event) {
    case ARDUINO_EVENT_WIFI_AP_START:
      Serial.print("AP Started: ");
      Serial.println(WiFi.softAPIP());
      delay(500);          // without delay, ipv6 replies did not work on AP iface, e.g. error code = -4
      WiFi.softAPenableIpV6();
      canStartWiFiServices();
      break;
    case ARDUINO_EVENT_WIFI_AP_STOP:
      Serial.println("AP Stopped");
      break;
    case ARDUINO_EVENT_WIFI_AP_GOT_IP6:
      Serial.print("AP IPv6: ");
      Serial.println(WiFi.softAPIPv6());
      canStartWiFiServices();
      break;
    case ARDUINO_EVENT_WIFI_STA_START:
      //WiFi.setHostname(ssid_ap);
      Serial.println("STA Started");
      break;
    case ARDUINO_EVENT_WIFI_STA_CONNECTED:
      Serial.println("STA Connected.");
      delay(1000);          // without delay, no ipv6 replies are sent via STA iface, e.g. error code = -4
      WiFi.enableIpV6();
      break;
    case ARDUINO_EVENT_WIFI_STA_GOT_IP6:
      Serial.print("STA IPv6: ");
      Serial.println(WiFi.localIPv6());
      canStartWiFiServices();
      break;
    case ARDUINO_EVENT_WIFI_STA_GOT_IP:
      Serial.print("STA IPv4: ");
      Serial.println(WiFi.localIP());
      canStartWiFiServices();
      break;
    case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:
      Serial.println("STA Disconnected");
      break;
    case ARDUINO_EVENT_WIFI_STA_STOP:
      Serial.println("STA Stopped");
      break;
    default:
      break;
  }
}

void canStartWiFiServices() {
  Serial.println("canstart");
  canstart = true;
  shallstart = true;
}

void startWiFiServices() {
  Serial.println("startWiFiServices");
  if(udp.listen(65000)) {
    Serial.println("UDP Listening.");
    udp.onPacket([](AsyncUDPPacket packet) {
        Serial.print("UDP ");
        Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
        Serial.print(" from: ");
        if (packet.isIPv6()) {
          Serial.print(packet.remoteIPv6());
        }
        else {
          Serial.print(packet.remoteIP());
        }
        Serial.print(":");
        Serial.print(packet.remotePort());
        Serial.print(" to: ");
        Serial.print(packet.interface());
        Serial.print(":");
        Serial.print(packet.localPort());
        Serial.print(", len: ");
        Serial.print(packet.length());
        Serial.println();
        if (packet.isIPv6()) {
          size_t res = udp.writeTo((const uint8_t*) "Reply6\r\n", 8, packet.remoteIPv6(), 5000, packet.interface());
          Serial.printf("res6 = %d (%d)\n", res, udp.lastErr());
        }
        else {
          size_t res = udp.writeTo((const uint8_t*) "Reply4\r\n", 8, packet.remoteIP(), 5000, packet.interface());
          Serial.printf("res4 = %d (%d)\n", res, udp.lastErr());
        }
    });
  }
  services = true;
}

void setup()
{
  WiFi.disconnect(true);
  Serial.begin(115200);
  Serial.println("\n\nConnecting.");
  WiFi.onEvent(WiFiEvent);
  WiFi.mode(WIFI_MODE_APSTA);
  WiFi.softAP(ssid_ap, pass_ap);
  WiFi.begin(ssid, password);
}

void loop()
{
  if (canstart && shallstart && !services) {
    startWiFiServices();
  }
  delay(100);
}

```

Python program:

```python
import socket

# listen on port 5000
udp = socket.socket(socket.AF_INET6, socket.SOCK_DGRAM)
udp.bind(('', 5000))

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)
sock.sendto(b'Test4\r\n', ("192.168.4.255", 65000))

sock6 = socket.socket(socket.AF_INET6, socket.SOCK_DGRAM)
ifidx = socket.if_nametoindex("wlp3s0")
sock6.setsockopt(socket.IPPROTO_IPV6, socket.IPV6_MULTICAST_IF, ifidx)
sock6.sendto(b'Test6\r\n', ("ff02::1", 65000))

udp.settimeout(5)
while True:
    try:
        data, addr = udp.recvfrom(4096)
        print(f"{addr}: {data}")
    except socket.timeout:
        break

```

### Debug Messages (no IP6 packet reply):
```
[    39][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   128][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
STA Started
[   129][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 10 - AP_START
AP Started: 192.168.4.1
canstart
startWiFiServices
UDP Listening.
[  2023][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 16 - AP_GOT_IP6
AP IPv6: fe80:0000:0000:0000:260a:c4ff:fec5:cb7d
canstart
[  2600][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
STA Connected.
[  5023][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 8 - STA_GOT_IP6
STA IPv6: fe80:0000:0000:0000:260a:c4ff:fec5:cb7c
canstart
[  5610][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[  7655][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
UDP Broadcast from: 192.168.4.2:50523 to: 1:65000, len: 7
res4 = 8 (0)
UDP Multicast from: fe80:0000:0000:0000:35fb:962e:14f4:821b:33800 to: 1:65000, len: 7
res6 = 8 (0)
```

Python output (last line might not be present :P ):
```
$ python udp_test_minimal.py 
('::ffff:192.168.4.1', 65000, 0, 0): b'Reply4\r\n'
('fe80::260a:c4ff:fec5:cb7d', 65000, 0, 3): b'Reply6\r\n'
```

fapolumi, 2021-11-03:
I just found out that changing the Debug Level to "Error" will result in IP6 packets being send, while setting it to "Debug" will result in the opposite behavior for me quite reproducible. This is by far the most reliable way to trigger this behavior at this point.

VojtechBartoska, 2022-04-07:
Hello @fapolumi, are you able to give this a try on v2.0.3-RC1? Thanks

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska