espressif/arduino-esp32 #5841: UpdateClass unfriends Platformio and Arduino IDE

tobozo, 2021-11-03:
Plaftorms: both platformio and Arduino IDE 1.8.16
Module: ESP32
Core version: 2.0.1-rc1
System: linux (ubuntu-latest)

Problem: binaries from platformio and Arduino IDE don't seem to be able to start each other using Update.h
As a result I can't SD-Load a pio binary when the ESP has been flashed from Arduino IDE (and vice-versa), and I can't use Update.rollBack() either when the origin platforms of the OTA partitions differ.

What I'm trying to do: produce binaries from Arduino IDE I can load (SD, OTA or rollBack) from platformio using Update.h library, and vice-versa.

I'm getting this after what appears a successful `Update.rollBack()` from a pio binary to an arduino-ide binary:

```log
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
�ets Jun  8 2016 00:22:57

then (forever loops on RTCWDT_RTC_RESET)

```

me-no-dev, 2021-11-04:
this is bizarre :D could it be that pio is building with older arduino? like before 2.0.0?

tobozo, 2021-11-04:
It's using `platform-espressif32#feature/arduino-upstream` for both 2.0.0 and 2.0.1-RC1, and only RC1+platformio binaries can demonstrate the issue.

This fails:

```ini
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1-RC1
```

This succeeds:

```ini
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.0
```

I've also had other issues with RC1+platformio, some conditional macros are ignored (e.g. `#ifdef _SD_H_`) depending on the nesting depth.

So I'm speculating on a possible missing compilation flag that appeared with 2.0.1-RC1 and isn't yet implemented on platformio.

Jason2866, 2021-11-04:
We have no problems building Tasmota, with the framework builded with the arduino lib builder (based on IDF4.4 branch and Arduino-esp32) with Platformio.
Issues you have are maybe from Platformio LDF (Library Dependency Finder).
We had to change our setup after a small code change. LDF did weird...

tobozo, 2021-11-04:
@Jason2866 thanks for your feedback, I tried to pick some build_board settings from tasmota32 earlier but most are ignored since this issue only appears to affect `framework = arduino`, also tasomta32 seems to use `platform = espressif32 @ 3.3.1`.

I found out `espressif32 @ 3.3.1` can't be used with `arduino-esp32.git#2.0.1-RC1` or even 2.0.0 (pio complains about some i2c refs missing), only `platform-espressif32.git#feature/arduino-upstream` does not complain.

So this yields more suspicion towards `#feature/arduino-upstream`.

Jason2866, 2021-11-04:
@tobozo we use this without any issue
```
[env:tasmota32idf4]
extends                 = env:tasmota32_base
platform                = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages       = framework-arduinoespressif32 @ https://github.com/tasmota/arduino-esp32/releases/download/2.0.1rc1/framework-arduinoespressif32-release_IDF4.4.tar.gz
                          platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
                          platformio/tool-mklittlefs @ ~1.203.200522
build_unflags           = ${env:tasmota32_base.build_unflags}
                          -Wswitch-unreachable
                          -Wstringop-overflow
                          -Wincompatible-pointer-types
build_flags             = ${env:tasmota32_base.build_flags}
                          -Wno-switch-unreachable
                          -Wno-stringop-overflow
```

tobozo, 2021-11-04:
`#feature/arduino-idf-master` isn't a very happy build, although it works fine with esp-idf, it fails with `framework = arduino` with different compilation errors.

This issue is just for the sakes or reporting an edgy case anyway, I'm secretly hoping that plaformio will soon release an updated `platform-espressif32` package so I can take this exception out of my mind :-)

[image]

Jason2866, 2021-11-04:
Weird build errors can have the reasons in platformio setup.
I invested already "some" hours to solve issues which came out of the dark.
Do not make bets that a official release will solve. There will be not really a difference.
Anyway without Platformio complex projects would not be possible!!

tobozo, 2021-11-04:
Tasmota repository is exactly where my attention was before I created this issue, very nice work and thorough use of the ini files!

VojtechBartoska, 2022-04-07:
Hi @tobozo, I think we can close this one as expired, right?

tobozo, 2022-04-07:
closing this a solved (by using @tasmota's [platform](https://github.com/tasmota/platform-espressif32) :bow: )

closed 2022-04-07 by tobozo

Jason2866, 2022-04-07:
@tobozo Looking at https://github.com/Jason2866/platform-espressif32/actions/runs/2090321079
there is all green