espressif/arduino-esp32 #5843: BLE Device: disconnect all existing connections when only one device disconnected

tientu0001, 2021-11-03:
Hardware:
Board: ESP32 Dev Module
IDE name: Arduino IDE
Flash Frequency: 80Mhz
OS: Windows 10

Description:
I use example BLE_Client and custom, now my ble client will connect to 3 ble server. All works excellent
But when one ble server disconnected, so all all existed connections was disconnected too.
[image]

P-R-O-C-H-Y, 2021-11-04:
Hi, please can you add more info from issue template?
`Arduino-esp32 version`
`Sample of code / sketch`
Thanks

P-R-O-C-H-Y, 2021-11-15:
@tientu0001 Can you add more info that I asked before?

closed 2021-11-17 by tientu0001