espressif/arduino-esp32 #5848: assertion "pbuf_free: p->ref > 0" failed with MQTT

garudaonekh, 2021-11-04:
pbuf_free exception after a few MQTT request/response

This error happen after a few MQTT request/response.

"ERROR Message"
assertion "pbuf_free: p->ref > 0" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c", line 766, function: pbuf_free
abort() was called at PC 0x4010df83 on core 0

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
[ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

```
Exception Cause: Not found

0x4010df83: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x400892dc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x40089559: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4010df83: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x4014fefb: pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:1332
0x4014fc79: tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:504
0x40155ffb: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:750
0x4015a8c6: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:184
0x40144fa3: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
0x4008a56a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
0x40078000: ?? ??:0
0x40080400: ?? ??:0
0x400806a8: ?? ??:0
```

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>ESP32 Dev Module
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B> Platform.io
|<B>Flash Frequency</B>|40Mhz|80Mhz|
|<B>PSRAM enabled</B>|yes|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>Mac OSX

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me-no-dev, 2021-11-04:
please provide example code.

garudaonekh, 2021-11-04:
> please provide example code.

```
static void packet_receive_handler(void* handler_args, esp_event_base_t base, int32_t id, void* event_data)
{   
    
    struct pbuf *pb = *(struct pbuf **)event_data;
    int errnum;
    if (pb == NULL)
        return;
    if (mqtt_if->netif.input(pb, &mqtt_if->netif) != ERR_OK)
    {
        
        ESP_LOGI(TAG, "input failed");
        pbuf_free(pb);
    }
}
```

After my further investigation, it happened during incoming MQTT message processing. Sorry, this C style stuff coding is foreign to me.

me-no-dev, 2021-11-04:
in your code you free the pbuf only when there is an error. you should free it in all cases, once you have nothing more to do with it.

closed 2021-11-04 by me-no-dev

garudaonekh, 2021-11-05:
> in your code you free the pbuf only when there is an error. you should free it in all cases, once you have nothing more to do with it.

That doesn't solve my problem. Based on the code, pb is to be consumed and freed by netif.input function if there's no error so no need to free it if no error and it'll throw pbuf->free exception if I free it again.

The cause of this error(pbuf_free: p->ref > 0) is due to reference count is zero unintentionally.

me-no-dev, 2021-11-05:
reference count is NOT zero is the problem, which means that the buf is referenced somewhere but not freed. pbuf_free subtracts one reference and if zero, then it really frees the buffer. That is an issue with the library, not with Arduino

garudaonekh, 2021-11-08:
Sorry for coming again. I still can't figure out what went wrong because it happened randomly when MQTT packet is received. However, it points to this line of code: **if (mqtt_if->netif.input(pb, &mqtt_if->netif) != ERR_OK)**

netif.input() belongs to netif API class of this arduino-esp32.

Thus based on your comment, it means pb->ref should be zero, otherwise error will happen?

```
static void packet_receive_handler(void* handler_args, esp_event_base_t base, int32_t id, void* event_data)
{   
    
    struct pbuf *pb = *(struct pbuf **)event_data;
    int errnum;
    if (pb == NULL)
        return;
    **if (mqtt_if->netif.input(pb, &mqtt_if->netif) != ERR_OK)**
    {
        
        ESP_LOGI(TAG, "input failed");
        pbuf_free(pb);
    }
}

```

me-no-dev, 2021-11-08:
while we do provide it it is not a part of Arduino. It's part of LwIP that is compiled with ESP-IDF to be used by Arduino.
There are a couple more places in that mqtt lib that creates pbufs and refs them. May be that is where is the problem?
And yes, the error says that `pbuf->ref` was not zero when the stack tried to free the pbuf.