espressif/arduino-esp32 #5849: ESP32 dowd v3 using CH340K error upload

qiwalker, 2021-11-05:
hi 
i try to build my own esp32 for serial com i use CH340K driver already installed and com port detected
when i try to upload using arduino IDE board choose esp32 dev module
error appear

esptool.py v3.0-dev
Serial port COM6
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 42.51MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 78:e3:6d:e9:cf:74
Uploading stub...
Traceback (most recent call last):
  File "esptool.py", line 3682, in <module>
  File "esptool.py", line 3675, in _main
  File "esptool.py", line 3353, in main
  File "esptool.py", line 716, in run_stub
  File "esptool.py", line 605, in mem_block
  File "esptool.py", line 406, in check_command
  File "esptool.py", line 369, in command
  File "esptool.py", line 332, in write
  File "site-packages\serial\serialwin32.py", line 323, in write
serial.serialutil.SerialTimeoutException: Write timeout
Failed to execute script esptool
An error occurred while uploading the sketch

how to solve this problem
thanks
rizki

me-no-dev, 2021-11-05:
what baudrate are you using to upload? those CH chips are a bit shitty and do not upload fine with high baudrates. Can you try 115200?

qiwalker, 2021-11-05:
after i check with multimeter the TX pin on CH340K is 4.2VDC and the RX pin is 3.2VDC
the CP2021 chip RX and TX is 3.2 VDC
will it the cause of my problem ?

VojtechBartoska, 2022-02-15:
@qiwalker Hi, is your issue still unresolved?

VojtechBartoska, 2022-03-03:
Hi, I'm closing the issue due to no resposne. If needed you can reopen it. Thanks for your contribution!

closed 2022-03-03 by VojtechBartoska