espressif/arduino-esp32 #5850: No IPv6 for lower debug levels without extra delays in "WiFiIPv6" example

fapolumi, 2021-11-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			        2.0.0
IDE name:						Arduino IDE 1.6.18
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                          Linux

### Description:
Hi everyone, I'm having trouble with the WiFiIPv6.ino example.

To reliably get an IPv6 STA address, I have to add a delay() just before "enableIPv6()" in line 82. Am I doing something wrong? Is there any documentation for that? I tried different APs (e.g. FritzBox/hostapd) with the same results.

I'm using a plain ESP32 devboard, with nothing else connected. Even changing the debug level has an impact on the probability of getting an IPv6 address...

To complicate things, even if I get an IPv6 address, that might not be enough to reliably send packets. (separate issue)

Link to example: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino

Thanks for your time!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
see https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino

### Debug Messages:

Enabling "Debug" level will lead to my ESP getting an IPv6 address in 9/10 tries. Setting it to "Error" reduces the probability to around 20%. This log shows the less probable fault, that is, no IPv6 but "Debug" enabled.

```
[    38][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   124][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
[   130][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 10 - AP_START
[   138][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 11 - AP_STOP
[   138][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 10 - AP_START
[  2022][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 16 - AP_GOT_IP6
AP IPv6: fe80:0000:0000:0000:260a:c4ff:fec5:cb7d
[  2609][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  2626][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  2626][D][WiFiGeneric.cpp:857] _eventCallback(): STA IP: 192.168.178.51, MASK: 255.255.255.0, GW: 192.168.178.1
STA Connected
STA IPv4: 192.168.178.51
UTC: 07:56:14 (GMT)
UTC: 07:56:24 (GMT)
```

fapolumi, 2021-11-06:
I just checked, enableIPV6() returns false during faulty runs.

VojtechBartoska, 2022-04-07:
Same here as in your another issue, still valid on v2.0.3-rc1?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska