espressif/arduino-esp32 #5852: reading error SPI  value when connect Wifi

dungdao191299, 2021-11-05:
I'm trying use ESP32 read SPI from ADC module  to read analog value with high resolution and high sampling rate (about 5000 samples in 1s) for my energy monitoring project. After that i use WiFi to publish payload via MQTT. But after connected to WiFi, ESP32 received error value:
[image]
I'm pretty sure the reason is at the Wifi library because if i don't connect wifi, ESP32 received correct value:
[image]

atanisoft, 2021-11-05:
Pretty sure this will not get anywhere as there is no code example or steps to reproduce.

VojtechBartoska, 2021-12-14:
@dungdao191299 Can you please provide more info?

closed 2022-01-27 by VojtechBartoska