espressif/arduino-esp32 #5858: ESP32 WROVER-B 16MB HIMEM fails

marchingband, 2021-11-07:
using arduino-esp32 version 1.6 installed through boards manager
Arduino IDE version 1.8.12
MAC OSX 10.13.6

I choose the board `ESP32 Wrover Module` 
There is no option to enable PSRAM in the menu

```
#include <esp_spiram.h>
#include <esp_himem.h>

void setup() {
    Serial.begin(115200);
    printf("himem free %u\n", esp_himem_get_free_size());
    printf("himem phys %u\n", esp_himem_get_phys_size());
    printf("himem reserved %u\n", esp_himem_reserved_area_size());
}
```

the compiler confirms `Auto-detected Flash size: 16MB`

in the log I get : 
```
E (126) spiram: SPI RAM not initialized
E (126) esp_himem: Cannot allocate memory for meta info. Not initializing!
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
himem free 0
himem phys 4456448
himem reserved 262144
```

if I choose `ESP32 Dev Module`
`PSRAM` appears in the menu and I select `enabled`

compiler confirms again `Auto-detected Flash size: 16MB`

and the log says 
```
E (125) spiram: SPI RAM not initialized
E (125) esp_himem: Cannot allocate memory for meta info. Not initializing!
himem free 0
himem phys 4456448
himem reserved 262144
```

this is the same result as reported by other in issues like #4625 #4116
In a response in #5857 @me-no-dev said it should work, I must be missing something? Perhaps it works in arduino-esp32 v2.0beta?

chegewara, 2021-11-08:
HIMEM is not for flash, it is for PSRAM usage.

marchingband, 2021-11-08:
@chegewara ok that clears that up a little thank you. Regardless WROVER-B has 8MB PSRAM, and the initialization of it is failing here.

marchingband, 2021-11-08:
The part is labeled ESP-PSRAM64H ... 8MB

chegewara, 2021-11-08:
Did you enable psram in tools menu?

marchingband, 2021-11-08:
@chegewara yes

chegewara, 2021-11-08:
Ok, do you have something connected to pins GPIO16 and/or GPIO17? Those are psram pins.

marchingband, 2021-11-08:
No I don't.
ps_malloc does work, but only has the lower 4MB, so PSRAM is working but the himem is not

marchingband, 2021-11-08:
@chegewara 
using arduino-esp32 version 1.6 installed through boards manager
Arduino IDE version 1.8.12
MAC OSX 10.13.6

selected ESP32 Dev Board in the tools menu 
(I also tried ESP32 Wrover Module, though the PSRAM ENABLE option does not appear in the menu)

me-no-dev, 2021-11-08:
I can not explain why PSRAM init fails for you. You can try 2.0.0, but that might not help. Sounds a bit like hardware issue?

marchingband, 2021-11-08:
@me-no-dev @chegewara 
I am confused because  #4625 and #4116 have the same problem and those issues were closed as stale.
They both end up referencing 3rd party libraries that use 'magical' work arounds.

marchingband, 2021-11-08:
Unfortunately I havn't been able to get 2.0 to work, so I can't test that, I am stuck on OSX 10.13 and it seems incompatible.

VojtechBartoska, 2021-11-08:
@marchingband yes, you need to update to MacOS 10.15. Check [here](https://github.com/espressif/arduino-esp32/issues/5639#issuecomment-949618260)

marchingband, 2021-11-08:
@me-no-dev I don't see how it could be hardware, because `ps_malloc()` works perfectly, or am I missing something?
in #4116 @chegewara and @Nirvana-56 seem to get it working under Arduino-as-component, but not under Arduino. Then @me-no-dev is asked to rebuild Arduino with some flags enabled.
Did this happen? If it's possible for you to direct me to that commit, maybe I can track down the problem?

marchingband, 2021-11-08:
@me-no-dev I am on esp-idf 3.3 for example, maybe this is a mismatch?

marchingband, 2021-11-16:
@me-no-dev @chegewara any ideas?
Can we mark this as a bug?
I can't be the only person who needs the himem API to work...

PilnyTomas, 2022-03-07:
Hi, @marchingband 
Usage of SPI RAM is disabled when building default libs for the Arduino ESP32 core. We can not enable them by default because it would cause problems for some modules.
There are two options - use ESP-IDF or build your own Arduino ESP32 core.

1. If you need to use Arduino you will have to recompile libs with [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) with sdkconfig parameter changed.

  - Clone [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder)
  - open `sdkconfig.esp32` and find line containing `CONFIG_SPIRAM_BOOT_INIT`
  - change the line from comment to `CONFIG_SPIRAM_BOOT_INIT=y`
  - open `build.sh` and uncomment the last line `./tools/copy-to-arduino.sh`
  - run `./build.sh`
 Now you should have new libs in your Arduino folder compiled with the desired parameter and your problem should be fixed.

2. If you are familiar with ESP-IDF and/or you don't necessarily need to use Arduino-ESP32

  - Create IDF project with your code see [Build System](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html) for more info.
  - In your project use `idf.py menuconfig` to navigate to the parameter: `Component config > ESP32-specific > Support for external, SPI-connected RAM` if this option is disabled (marked with `[ ]`) press Enter. New option below will appear
  - Enable option `SPI RAM Config > Enable bank switching for  >4MiB external RAM`
  - Set parameter `Amount of 32K pages to reserve for bank switching` to (4)
  - save and compile
  The himem should work now.

Hope this will help.

marchingband, 2022-03-07:
>    1. If you are familiar with ESP-IDF and/or you don't necessarily need

As a library maintainer, how can I then expose that custom core to my
Arduino users?

marchingband, 2022-03-07:
@PilnyTomas I know that some sdkconfig vars are available as command line arguments to the Arduino compiler (such as log level), why can't all of them be set in this way? Using all the available hardware on the board seems like it would be a priority.

me-no-dev, 2022-03-07:
@marchingband could you give an example of such sdkconfig variable? The only things you can control at runtime are things that are implemented in the core itself. Everything from IDF is already precompiled with the options in sdkconfig.

me-no-dev, 2022-03-08:
all examples that you are mentioning either have to do with final linking or are otherwise implemented in Arduino. Flash options define which of the 8 prebuilt bootloaders is to be used.

To spin up your own install you need to fork this repo and the lib-builder, recompile the libs with the necessary options enabled and then import the resulting libs into your fork.

marchingband, 2022-03-08:
@me-no-dev 

> all examples that you are mentioning either have to do with final linking or are otherwise implemented in Arduino. Flash options define which of the 8 prebuilt bootloaders is to be used.
> 

I am aware if this, that's why I am requesting we need more such prebuilt binaries to accommodate users who want to use the himem. Adding one more dimension of binaries, and an Arduino menu option for himem use.

> To spin up your own install you need to fork this repo and the lib-builder, recompile the libs with the necessary options enabled and then import the resulting libs into your fork.

thanks, I will try this. So i assume I can then distribute this custom fork to my users via the Arduino board file system? Is there an example or documentation of this process available?

me-no-dev, 2022-03-09:
himem is not an often requested feature. You are either the first one or the second person who asked for it. Adding more libraries and combinations is an issue. It adds space to the repository.

You can check the build scripts in this repo for release and see how everything is packaged for consumption by the board manager in Arduino: https://github.com/espressif/arduino-esp32/blob/master/.github/scripts/on-release.sh

closed 2022-03-16 by VojtechBartoska