espressif/arduino-esp32 #5859: C++20 for ESP32?

VioletGiraffe, 2021-11-08:
This is not a bug report, more of a request / question. Is there a way to get C++20 working? And what's the latest version of `toolchain-xtensa32`? I'm using `3.80200.200512`, this one recognizes `-std=gnu++2a`, but it's GCC 9 - some important features are missing, like `consteval`. Is there a ESP32 toolchain with GCC 10? Or, at least, is it being worked on?

igrr, 2021-11-08:
This is being worked on, we will switch to GCC 11 in the next toolchain release. The tracking issue is https://github.com/espressif/esp-idf/pull/4846

VioletGiraffe, 2021-11-08:
@igrr, that's great news, thank you!

closed 2021-11-08 by VioletGiraffe

hansmbakker, 2023-01-28:
Work is being done in https://github.com/espressif/arduino-esp32/pull/7733