espressif/arduino-esp32 #5862: deepsleep: which wakeup-source is currently monitored?

Paulsburg, 2021-11-09:
Hi,
I am programming esp32 with the arduino IDE.

in my sketch, I am using different modes of deep sleep: ext0 ext1 and RTC-Timer as sources. Depending of the program-needs.

Do I can test in my sketch which deep sleep wakeup-mode (ext0 or ext1) currently is activ?

I am searching a bug in my sketch and for test purposes I want the print out the current  programed wakeup-mode/source .

To be clear, I don´t want to print out the current wakeup_source after a wakeup. I am interested to know which of the possible sources ext0 or ext1 would be active monitored in deep sleep mode.

Hope, my question is clear... :-(

Thanks for your help

felmue, 2021-11-09:
Hello @Paulsburg 

ok, if I understand correctly you'd like to call `esp_sleep_enable_X_wakeup()` in various places in your code and then just before the ESP32 is put to deep sleep you'd like to know which wake up sources your code actually has enabled?

Reading through the Espressif documentation I don't see a function for that.

That said, I suggest that in your sketch whenever you call an `esp_sleep_enable_X_wakeup()` function you also note that information into a global variable for later reference.

Thanks
Felix

Paulsburg, 2021-11-09:
Hi Felix

thats such a simple solution.... Sometimes to simple to see it ;-)

Thanks for your advice/help
Axel

closed 2021-11-09 by Paulsburg