espressif/arduino-esp32 #5863: compile error: ESP_TASK_MAIN_PRIO was not declared in this scope

dsyleixa, 2021-11-09:
Arduino IDE 1.8.9, Adafruit Feather ESP32, ESP32 core 1.0.6:

Formerly I was using 
vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO );
in my programs, but now after migrating to a new Notebook I suddenly get the compile error
`'ESP_TASK_MAIN_PRIO' was not declared in this scope`

OTOH, changing ESP_TASK_MAIN_PRIO e.g. to numbers like 0 or 1 all is running like a charm.

what am I missing?

```
Arduino: 1.8.9 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None, Default"

D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\helmu\AppData\Local\Temp\arduino_build_8753 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.0.0 -verbose D:\Akten\Programmierung\ArduinoProgs\Test\std_thread\std_thread_000\std_thread_000.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware D:\Arduino\portable\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools D:\Arduino\portable\packages -built-in-libraries D:\Arduino\libraries -libraries D:\Arduino\portable\sketchbook\libraries -fqbn=esp32:esp32:featheresp32:FlashFreq=80,UploadSpeed=921600,DebugLevel=none,PartitionScheme=default -ide-version=10809 -build-path C:\Users\helmu\AppData\Local\Temp\arduino_build_8753 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\Arduino\portable\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=D:\Arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=D:\Arduino\portable\packages\esp32\tools\esptool_py\3.0.0 -verbose D:\Akten\Programmierung\ArduinoProgs\Test\std_thread\std_thread_000\std_thread_000.ino
Using board 'featheresp32' from platform in folder: D:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6
Using core 'esp32' from platform in folder: D:\Arduino\portable\packages\esp32\hardware\esp32\1.0.6
cmd /c if exist "D:\\Akten\\Programmierung\\ArduinoProgs\\Test\\std_thread\\std_thread_000\\partitions.csv" copy /y "D:\\Akten\\Programmierung\\ArduinoProgs\\Test\\std_thread\\std_thread_000\\partitions.csv" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_8753\\partitions.csv"
cmd /c if not exist "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_8753\\partitions.csv" copy "D:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6\\tools\\partitions\\default.csv" "C:\\Users\\helmu\\AppData\\Local\\Temp\\arduino_build_8753\\partitions.csv"
        1 Datei(en) kopiert.

Detecting libraries used...
"D:\\Arduino\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/config" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_trace" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/app_update" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/asio" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bootloader_support" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/bt" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/coap" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/console" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/driver" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/efuse" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp-tls" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp32" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_adc_cal" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_event" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_client" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_http_server" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_ota" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_https_server" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_ringbuf" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/esp_websocket_client" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/espcoredump" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/ethernet" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/expat" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/fatfs" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freemodbus" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/freertos" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/heap" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/idf_test" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/jsmn" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/json" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/libsodium" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/log" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/lwip" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mbedtls" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mdns" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/micro-ecc" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/mqtt" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/newlib" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nghttp" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/nvs_flash" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/openssl" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/protobuf-c" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/protocomm" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/pthread" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0.6/tools/sdk/include/sdmmc" "-ID:\\Arduino\\portable\\packages\\esp32\\hardware\\esp32\\1.0 [... cut ...]
std_thread_000:24:28: error: 'ESP_TASK_MAIN_PRIO' was not declared in this scope

    vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio   

                            ^

D:\Akten\Programmierung\ArduinoProgs\Test\std_thread\std_thread_000\std_thread_000.ino: In function 'void loop0()':

std_thread_000:34:28: error: 'ESP_TASK_MAIN_PRIO' was not declared in this scope

    vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio

                            ^

exit status 1
'ESP_TASK_MAIN_PRIO' was not declared in this scope
```

me-no-dev, 2021-11-09:
Probably missing include: https://github.com/espressif/arduino-esp32/blob/00214d5c2a1c2b1904f2caf6f0d5ddfe952331ff/tools/sdk/esp32s2/include/esp_system/include/esp_task.h#L62

closed 2021-11-09 by me-no-dev

dsyleixa, 2021-11-09:
I never had additional includes other than the ones I wrote above.
```
#include <Arduino.h>
#include <thread>
#include <freertos/task.h>
```
So as this issue is still unsoveld, which `#include` is needed now additionally?

me-no-dev, 2021-11-09:
this issue has nothing to do with Arduino. You are using some ESP-IDF defines. Just include the correct header...

dsyleixa, 2021-11-09:
it's an error thrown by my Arduino IDE, so to me it's an esp-arduino-issue. Finally as an Arduino-user I have no idea in which nested package it's packed in.
So after all I assume that you might mean
`#include <esp_task.h>`
Is that new? 
when has that been changed?
But for now, although you didn't advice it explicitely, and as it turns out it could fix it, thanks.