espressif/arduino-esp32 #5864: ESP_ARDUINO_VERSION_PATCH not updated by build script in 2.0.1-RC1 and 2.0.1

tobozo, 2021-11-09:
Hello,

The new `ESP_ARDUINO_VERSION ` macro can't be used yet to distinguish 2.0.0 from 2.0.1-RC1 and 2.0.1.

Something is probably missing in the build scripts to increment the values.

[image]

[image]

Tomat7, 2021-12-23:
The same issue on 2.0.2 :-(

VojtechBartoska, 2021-12-23:
Hi, @Tomat7 and @tobozo we will fix this in 2.0.3. Adding the milestone.

tobozo, 2021-12-24:
@me-no-dev @VojtechBartoska 

it's remotely related but none of the bundled libraries that were recently modified had their version number increased in their respective library.properties file either 

e.g. SD.h and Wire.h are still at 2.0.0

me-no-dev, 2021-12-25:
@tobozo are those versions used somewhere?

tobozo, 2021-12-25:
@me-no-dev those version numbers are visible in the Arduino IDE compilation recap so there's at least an added value for debugging.

It's also a good signal when one of the bundled libraries has changed e.g. SimpleBLE is unchanged since arduino-esp32-2.0.0 whereas BlueToothSerial received changes for arduino-esp32-2.0.2, would it make sense to synchronize the version numbers of the modified libraries with their respective arduino-esp320xxx version instead of using semver?

I know  platformio can to pick one particular version, for example it made the difference when SD.h was bumped to 2.0.0 even though no code in the library was changed.

drws, 2022-02-23:
@VojtechBartoska If you're fixing this one, please also consider the related and partly overlapping #5536. It is currently tagged for 2.1.0, but it makes sense to fix these two at once. Thanks!

VojtechBartoska, 2022-04-06:
This issue was closed by merging S3 support into master branch.

closed 2022-04-06 by VojtechBartoska