espressif/arduino-esp32 #5868: ESP32 Core v2.0.1 / 2.0.1 RC1 crashes if using pin pinMode with GPIO1

khoih-prog, 2021-11-10:
Core ESP32 Core **v2.0.1 / 2.0.1 RC1** crashes if using pin pinMode with GPIO1, INPUT or OUTPUT mode. No issue with core **v2.0.0-**

With the following simple sketch, it's working OK if using `PIN_D2` or `PIN_D3`, but crash if using `PIN_D1`

```
#define PIN_D1            1         // Pin D1 mapped to pin GPIO1/TX0 of ESP32
#define PIN_D2            2         // Pin D2 mapped to pin GPIO2/ADC12/TOUCH2/LED_BUILTIN of ESP32
#define PIN_D3            3         // Pin D3 mapped to pin GPIO3/RX0 of ESP32

#define PIN_OUT           PIN_D1

void setup() 
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);

  delay(1000);
  
  Serial.print("\nStarting D1_Crash using with pin = "); Serial.print(PIN_OUT); Serial.print(" on "); Serial.println(ARDUINO_BOARD);

  pinMode(PIN_OUT, OUTPUT);
}

void loop() 
{
  digitalWrite(PIN_OUT, HIGH);
  Serial.println("H");        // H means High
  delay(1000);
  digitalWrite(PIN_OUT, LOW);
  Serial.println("L");        // H means High
  delay(1000);
}
```

me-no-dev, 2021-11-10:
any logs from that crash?

khoih-prog, 2021-11-10:
Even with `Core Debug Level: "Verbose"`, the terminal debug has almost nothing

```
⸮
```

Using 

```
#define PIN_OUT           PIN_D2
```

the debug terminal is

```
Starting D1_Crash using with pin = 2 on ESP32_DEV
H
L
H
L
H
L
H
L
H
...
```

VojtechBartoska, 2022-04-07:
Hello, is this still valid @khoih-prog?

GPIO was refactored in meantime, if yes, are you able to test this on v2.0.3-RC1?

khoih-prog, 2022-04-07:
Hi @VojtechBartoska 

Just retest and still have crashes when using **PIN_D1** and **v2.0.3-RC1**

P-R-O-C-H-Y, 2022-04-13:
Hi @khoih-prog,
what board are you using with ESP32_DEV?

P-R-O-C-H-Y, 2022-04-13:
@khoih-prog 
For ESP32 the Serial uses by default these pins:
UART0 RX pin 3
UART0 TX pin 1 
Thats when you do pinMode on pin 1, the Serial no longer works.

khoih-prog, 2022-04-13:
Hi @P-R-O-C-H-Y 

Thanks for the clarification. So sorry, this is my **weird and funny** mistake. The board is still working, not crash at all. Just the Serial @ UART0 in not working.

closed 2022-04-13 by khoih-prog