espressif/arduino-esp32 #5869: Limit on total number of BLE services

DhirajGehlot, 2021-11-10:
Hello,

### Hardware:
Board:							ESP32-WROOM-32E
Core Installation version:			1.0.6
IDE name:						Arduino IDE?
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Mac OSX
Arduino version:                                    Arduino 1.8.15 

I am using Arduino IDE to code ESP-32 module. Currently I am using BLE functionality to communicate with our mobile app.
There seems to be limitation on number of BLE services which can be created. Even If I create more than 7 services, only 6 are visible (viewed in nrfConnect app : mobile application by nordic).

I find some link which mentions same issue as mine and have stated solution in ESP-IDF but I couldn't find the solution on Arduino platform:
https://github.com/espressif/esp-idf/issues/1644
https://github.com/espressif/esp-idf/issues/5495

_BLE STACK uses GATT_MAX_SR_PROFILES to create the number of gatt servers now, the default value is 8.
BLE STACK will also create GAP SERVICE itself, leaving the actual number of users to 6.
If you want to create more services, you can increase this macro definition as needed_

I am unable to find this file in Arduino package folder, Can someone help me where can I find this MACRO definition for Arduino platform.?

Thanks

me-no-dev, 2021-11-10:
All IDF menuconfig settings are preconfigured in Arduino and can not be changed, except if Arduino is used as IDF component. Then you can configure everything in IDF.

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L354

DhirajGehlot, 2021-11-10:
Thanks for the reply @me-no-dev 

_All IDF menuconfig settings are preconfigured in Arduino and can not be changed_
- So maximum BLE service are limited in Arduino? Also this config file is generated automatically? 

I navigated to my sdkconfig.h file and couldn't find this "CONFIG_BT_GATT_MAX_SR_PROFILES=8"
Can I add this variable CONFIG_BT_GATT_MAX_SR_PROFILES=8 manually in my SDK config file?

me-no-dev, 2021-11-11:
What do you mean by "your" sdkconfig file? The sdkconfigs used by arduino are located in `tools/sdk/[chip]/` and the header is in the `include/config` folder under that. BTW I advise you to use version 2.0.1 of this repository :)

Jason2866, 2021-11-22:
@DhirajGehlot The file in Arduino is Info what values are configured. You cant change settings here. If you want to change you have to use Arduino as IDF component or build your own Arduinio framework with https://github.com/espressif/esp32-arduino-lib-builder

VojtechBartoska, 2022-04-07:
@DhirajGehlot was your issue fully covered? I think we can close it, thanks!

closed 2022-04-07 by VojtechBartoska