espressif/arduino-esp32 #5871: Rollback with Arduino as ESP-IDF Component

GKFernandes, 2021-11-10:
Hello everyone,

I ran into the same issue as stated in: espressif/arduino-esp32#3318 and also in espressif/arduino-esp32#3749

I think this issue about the weakly defined verify0ta() should be fixed and made in such a way that the user can perform the verification later in the code instead of inside the initArduino() function.

me-no-dev, 2022-05-13:
How about we add a new weak function `bool verifyRollbackLater()` to allow you to skip verification in `initArduino` and do it yourself later?

GKFernandes, 2022-05-19:
Hello @me-no-dev , I think this could be a solution, but can the function verifyOta() just be removed?

What do you think @atanisoft @marcovannoord?

marcovannoord, 2022-05-19:
Yeah that would be a perfect solution i think, and it would just skip the verifyOTA() function.
In that case, if users want to use the verifyOTA to immediately verify rollback, then they can still use that function if they override the weak function.

me-no-dev, 2022-05-19:
if you overwrite `verifyRollbackLater`, then `verifyOTA` will not be called at all. This whole block will not execute: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L215-L226

marcovannoord, 2022-05-19:
Yeah i think that's the perfect solution. That way, users can add this feature, while keeping legacy things intact.

me-no-dev, 2022-05-19:
Done :)

closed 2022-05-19 by me-no-dev

zekageri, 2022-11-01:
Any example for this?