espressif/arduino-esp32 #5874: what should the ULP power consumptio be in deep sleep?

eyaleb, 2021-11-11:
I want to run my project from battery so need to optimize power usage.

The hardware is a naked esp-wroom-32 module soldered to a carrier board. A low Iq LDO (RT9080) is used with a Li-Ion battery. No devices are attached yet. The sw is built with IDF 4.4.

When I go into deep sleep the current is 10-12uA, as expected. If I activate the empty ULP program it uses 450uA. The doco suggests it should be around 150uA.

I use [this](https://github.com/makehackvoid/ESP8266Lib/tree/master/examples/eyal/null) project.

TIA

atanisoft, 2021-11-11:
@eyaleb the example project seems to be pure ESP-IDF and not arduino-esp32, I'd suggest asking over on https://github.com/espressif/esp-idf or even esp32.com in the forum.

eyaleb, 2021-11-11:
Oops, accidentally put it in the wrong forum.

closed 2021-11-11 by eyaleb