espressif/arduino-esp32 #5876: HardwareSerial::flush freezing ESP32

ShenRen, 2021-11-11:
The use of the flush function is likely to trigger a task deadlock
IDF V4.3.1
[image]

SuGlider, 2021-11-11:
Hi @ShenRen 
This issue seems related to https://github.com/espressif/esp-idf/issues/5156
The issue with ```uart_wait_tx_done()``` (used in ```HardwareSerial::flush()``` and then in ```uartFlushTxOnly()```) relates to IDF versions before July 20th 2021, most probably present in IDF 4.3.

It seems fixed in IDF 4.4 as reported here https://github.com/espressif/esp-idf/issues/5156#issuecomment-888520132
based on commit https://github.com/espressif/esp-idf/commit/c801b3a18209ae707d3f8224b0734315cdc99db8

Arduino Core 2.0.0+ uses IDF 4.4 (latest master). 
Thus, I think you may need to apply the fix to your local copy of IDF or to upgrade to IDF 4.4 in order to solve this issue.

SuGlider, 2021-11-12:
@ShenRen 
I confirm that IDF 4.3 has this bug. It is fixed in IDF 4.4.

The lines below shall be removed from the code (```/components/driver/uart.c``` in IDF 4.3) in order to fix it:
https://github.com/espressif/esp-idf/blob/release/v4.3/components/driver/uart.c#L1080
https://github.com/espressif/esp-idf/blob/release/v4.3/components/driver/uart.c#L1094-L1096

Those are equivalent to aplying the fix to uart.c as in this commit:
https://github.com/espressif/esp-idf/commit/c801b3a18209ae707d3f8224b0734315cdc99db8

VojtechBartoska, 2021-11-18:
@ShenRen Up to the last comment of @SuGlider. Can I consider this as solved or do you face some more problems? :)

ShenRen, 2021-11-21:
Yes, the problem seems to have been solved. thank you.

closed 2021-11-21 by SuGlider

beachmiles, 2023-08-31:
This bug appears to have re-appeared in Arduino v2.0.11 based on ESP-IDF v4.4.5 on my esp32-c3 AirM2M_CORE_ESP32C3. For the USB CDC serial port running the **`Serial.flush()`** command after outputting text from the wifi scan causes the micro to freeze when the COM port is not connected. Removing the **`Serial.flush()`** command fixes the freeing issue.