espressif/arduino-esp32 #5878: CameraWebServer Example No

TokenThinker, 2021-11-11:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

TokenThinker, 2021-11-12:
I think the "fd_forward.h" and "fr_forward.h" aren't being found, when trying to include them

```
example_sketch_fd:7:10: fatal error: fd_forward.h: No such file or directory
 #include "fd_forward.h"
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
fd_forward.h: No such file or directory

```

Jason2866, 2021-11-12:
Correct since face defection is not included in the core 2.0.1 build

TokenThinker, 2021-11-13:
@Jason2866 is there a plan to implement it or is it gone for good? If so what does that timeline look like?

CameraWebServer example

```
#if CONFIG_ESP_FACE_DETECT_ENABLED

#include "fd_forward.h"

#if CONFIG_ESP_FACE_RECOGNITION_ENABLED
#include "fr_forward.h"
```

Jason2866, 2021-11-13:
This is a question for the maintainers. I just do a modified Arduino framework for Project Tasmota with the https://github.com/espressif/esp32-arduino-lib-builder and noticed in this process that there is no more a compatible  Face Detection version for IDF4.4 It is replaced by https://github.com/espressif/esp-dl which is not implemented in Arduino at the moment.
@me-no-dev Are there plans for?

me-no-dev, 2021-11-13:
@Jason2866 I also recently noticed this. Surely will take a look at it and fix it at some point :)

TokenThinker, 2021-11-13:
@me-no-dev - You are amazing! If you can let me know as I am building PoC based of example code and parts. Was shocked when it stopped working.

I guess for now I can downgrade to re-enable functionality?

@Jason2866 - Thank you for the info.

closed 2021-11-13 by TokenThinker

chepo92, 2021-12-21:
Sorry to revive this thread, I tried v2.0.1 and added `#define CONFIG_ESP_FACE_DETECT_ENABLED 1` to app_httpd.cpp, face detection works but it has no output in serial monitor, also, what does face recognition do? I press enroll face when a face is detected with a rectangle but I think it does nothing to enroll it.

Jason2866, 2021-12-21:
@chepo92 You have not used core v2.0.1 in this case, there is no `fd_forward.h` and `fr_forward.h` in core 2.0.x

chepo92, 2021-12-21:
> @chepo92 You have not used core v2.0.1 in this case, there is no `fd_forward.h` and `fr_forward.h` in core 2.0.x

You were right, i checked pio output and it is using `framework-arduinoespressif32 3.10006.210326 (1.0.6)`

Token-Thinker, 2023-07-19:
hey doubling back on this topic seeing if there is a way i can get involved to help. I would really like to see this feature or something similar in the newer version or a submodule for it. let me know how i can get involved.