espressif/arduino-esp32 #5879: pbuf->ref > 0 assertion in tcp_in.c of lwip

garudaonekh, 2021-11-11:
Why pbuf->ref is 0 in tcp_in.c's tcp_input function randomly after a few incoming MQTT message?
I have been struggling with this issue for 5 days and can't figure out how it happen.

I use a library to tunnne all MQTT message to HTTP request. Thus the library will convert MQTT message and post it to TCP/IP stack and the Web Server on the ESP32 will response to it. But it keeps crashing after several mqtt message. ([https://github.com/levkovigor/MQTT_VPN_ESP32]

The error happen inside the netif.input(pb, &mqtt_if->netif) of tcp_in.c of lwip. It returns ERR_OK and then crashed. I think this function error when it tried to free other pbuf variable NOT the pb that's passed to it.

```
static void packet_receive_handler(void* handler_args, esp_event_base_t base, int32_t id, void* event_data)
{
    struct pbuf *pb = *(struct pbuf **)event_data;
    if (pb == NULL)
        return;

    ESP_LOGD(TAG, "Buffer received - len: %d tot_len: %d", pb->len, pb->tot_len);
    if (mqtt_if->netif.input(pb, &mqtt_if->netif) != ERR_OK)
    {
        ESP_LOGI(TAG, "input failed");
        pbuf_free(pb);
    }
    //printf("End rcv: %d\n", xPortGetFreeHeapSize());
}
```

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
[ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

```
0x4010cef7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x400892dc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x40089559: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4010cef7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
0x4014be1b: pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:1332
0x4014bb99: tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:504
0x40151f1b: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:750
0x401567e6: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:184
0x40140ea3: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
0x4008a56a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|ESP32-S2-Saola|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>Platform.io + VSCode
|<B>Flash Frequency</B>40Mhz
|<B>PSRAM enabled</B>|yes|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>Mac OSX

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

atanisoft, 2021-11-12:
This appears to be a bug in the library you are using, it could be any of the following:

1. pbuf exhaustion.
2. pbuf use after free.
3. unknown bug in code not from arduino-esp32.

This should be investigated further by the library owner.

garudaonekh, 2021-11-12:
> This appears to be a bug in the library you are using, it could be any of the following:
> 
> 1. pbuf exhaustion.
> 2. pbuf use after free.
> 3. unknown bug in code not from arduino-esp32.
> 
> This should be investigated further by the library owner.

**1. pbuf exhaustion.**
I have checked the library's code carefully, i doesn't free that pbuf at all unless tcp_in.c return error response-based on my tracking, tcp_in response ERR_OK but it still complain pbuf->ref > 0 assertion. I also check the memory heap for the possiblity of out of memory but still many heap left.

Since it happened randomly, I think it's some concurrent issue due to threading, but I have no idea what and how this happen.

atanisoft, 2021-11-12:
@garudaonekh I'd suggest open an issue on the library and instrument their code to find where/why this is occurring. It's entirely possible the code is exhausting the pbuf pool (there is a limit) and it is overrunning the LwIP stack from being able to cleanup or process.

If you are just using that library for MQTT access, I'd suggest trying the ESP-IDF native MQTT client directly.

garudaonekh, 2021-11-14:
thanks, i already tried to tried to ping the library's owner but no feedback yet.

closed 2021-11-27 by garudaonekh