espressif/arduino-esp32 #5882: Blocking code on Core 0 slows I2C

raphael-bmec-co, 2021-11-13:
### Hardware:
Board:							ESP32-D0WD-V3 (revision 3) custom board.
Core Installation version:			1.0.5 AND 2.0.0
IDE name:						Platform.io with Clion AND Arduino IDE.
Flash Frequency:					40Mhz.
PSRAM enabled:                                     Either.
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I2C communication slows drastically when core 0 is blocked. The results of testing alternate cores are shown below. 

I am assuming that the I2C interrupts are handled on core 0 and/or that there is a task on core 0 that handles the I2C? I am aware that the issue would go away if the blocking code was moved to core 1. However, in the case of WiFi/BLE we have no control over this.

Why would the I2C interrupts not be running on the core from which Wire was begun? I have even tried editing the esp32-hal-i2c.c to increase the priority of the I2C interrupts from `ESP_INTR_FLAG_LOWMED` to forced `ESP_INTR_FLAG_LEVEL3` AND clearing the `ESP_INTR_FLAG_SHARED` flag.

Looking forward to some feedback. It would be great if we could move whatever is slowing down the I2C to core 1?

@stickbreaker maybe this is a simple one for you to answer?

> Core 0 blocked, I2c on core 1, arduino on core 1
> blocked 50050us
> unblocked 473us
> 
> Core 1 blocked, I2c on core 0, arduino on core 1
> Doesn't run
> 
> Core 0 blocked, I2c on core 1, arduino on core 0 
> Doesn't run
> 
> Core 1 blocked, I2c on core 0, arduino on core 0 
> blocked 481us
> unblocked 474us

### Sketch - you will need an I2C device to reproduce. However, even without a device the transmission time is drastically different with and without the block on core 0.
``` cpp
#include <soc/timer_group_struct.h>
#include <soc/timer_group_reg.h>

#include <Wire.h>

void core0Loop(void *pvParameters){
  log_i("cor01Loop started on core: %u", xPortGetCoreID());
  
  for (;;){
    TIMERG0.wdt_wprotect = TIMG_WDT_WKEY_VALUE;
    TIMERG0.wdt_feed = 1;
    TIMERG0.wdt_wprotect = 0;
  }
}

void core1Loop(void *pvParameters){
  log_i("core1Loop started on core: %u", xPortGetCoreID());
  
  if (!Wire.begin())
  {
    log_e("Failed to start I2C");
  }

  for (;;){

    for (int brightness=0; brightness<256; brightness++){

    Wire.beginTransmission(0b00111110);
    Wire.write(0x56);
    Wire.write(brightness);
    int64_t t0 = esp_timer_get_time();
    if(Wire.endTransmission() != I2C_ERROR_OK){
      log_e("TRANSMISSION ERROR");
    }
    log_i("Transmission time %lldus", esp_timer_get_time() - t0);
    delay(2);
    }
    delay(500);
  }
}

void setup() {
  
  Serial.begin(115200);

  xTaskCreatePinnedToCore(
    core0Loop
    ,  "core0Loop"
    ,  5000
    ,  NULL
    ,  6
    ,  NULL 
    ,  0);

  xTaskCreatePinnedToCore(
    core1Loop
    ,  "core1Loop"
    ,  5000
    ,  NULL
    ,  6
    ,  NULL 
    ,  1);

    vTaskDelete(nullptr);
}

void loop()
{
  vTaskDelete(nullptr);
}
```

VojtechBartoska, 2022-04-08:
Hello, can you please test this under 2.0.3-RC1?

raphael-bmec-co, 2022-04-13:
Hi @VojtechBartoska. We are working on it and will revert ASAP.

me-no-dev, 2022-04-27:
@VojtechBartoska this could be answered best by someone who is more intimate with the FreeRTOS scheduler in ESP-IDF. It is probably caused by the scheduler not being able to work properly and the test is a bit invalid, since it abuses the watchdog in order not to cause reboot.

VojtechBartoska, 2022-05-19:
@me-no-dev Are you able to run the sketch to reproduce the issue?

me-no-dev, 2022-05-19:
@VojtechBartoska I do not know what device is tested and I probably do not have it, so no way to exactly test it here.

raphael-bmec-co, 2022-05-20:
Results of testing on 2.0.3-RC1.

The fourth case is now blocked at priority 6 now which does not make a lot of sense to me.

> Core 0 blocked, I2C on core 1, arduino on core 1
> blocked 5002 us - 9004 us (core 0 task priority 6, core 1 task priority 6)
> blocked 7002 us - 9005 us (core 0 task priority 1, core 1 task priority 6)
> unblocked 325 us - 458 us
> 
> Core 1 blocked, I2C on core 0, arduino on core 1
> Does not run (core 0 task priority 6, core 1 task priority 6)
> blocked 7002 us - 12001 us (core 0 task priority 1, core 1 task priority 6)
> unblocked 325 us - 449 us
> 
> Core 0 blocked, I2C on core 1, arduino on core 0
> Does not run (core 0 task priority 6, core 1 task priority 6)
> blocked 7002 us - 10005 us (core 0 task priority 1, core 1 task priority 6)
> unblocked 325 us - 458 us 
> 
> Core 1 blocked, I2C on core 0, arduino on core 0
> Does not run (core 0 task priority 6, core 1 task priority 6)
> blocked 9002 us - 12002 us (core 0 task priority 1, core 1 task priority 6)
> unblocked 325 us - 449 us

VojtechBartoska, 2024-02-20:
I am closing this as expired, if needed you can reopen.

closed 2024-02-20 by VojtechBartoska