espressif/arduino-esp32 #5885: WiFiProv Not Working

rsmck, 2021-11-13:
### Hardware:
Board:							ESP32
Core Installation version:			2.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                   no
Upload Speed:					921600
Computer OS:                                       macOS 

### Description:
WiFiProv simply doesn't seem to work for me at all.

When using the sample Sketch https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino the BLE doesn't appear within the ESPRESSIF app.

On one occasion it appeared, but as Prov_123 despite a different value set in the `WiFiProv.beginProvision()` call.

Using another BLE explorer app like LightBlue showed the device advertising `prov-scan`, `prov-session`, `prov-config`, `prov-ver` but it seems impossible to actually provision using the ESP tools.

### Sketch: 

(The void SysProvEvent(arduino_event_t *sys_event) {} is as https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino)

```cpp
void setup() {
  Serial.begin(115200);
  WiFi.onEvent(SysProvEvent);
  WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, "ATLN3915", "HA123");
}
```

### Debug Messages:
```
[ HH⸮⸮0][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    21][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
[    21][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 33 - PROV_INIT
[    24][I][WiFiProv.cpp:142] beginProvision(): Starting AP using BLE. service_name : HA123, pop : ATLN3915
[   117][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
[   525][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 35 - PROV_START

Provisioning started
Give Credentials of your access point using " Android app "
```

However the app (tried both iOS and Android) doesn't see it at all.

coderforlife, 2021-11-28:
I am also having this issue on the ESP32-C3 chip. I have tried iOS. The LightBlue app asks for approval to use Bluetooth services while the ESP app does not. It seems the ESP app isn't using Bluetooth at all? This is sounding more like an issue to https://github.com/espressif/esp-idf-provisioning-ios/tree/master/ESPProvision

coderforlife, 2021-11-28:
Check out https://github.com/espressif/esp-idf-provisioning-ios/issues/35 (at least for iOS).

VojtechBartoska, 2022-04-07:
Hello, can you please test this with 2.0.3-RC1 version?

closed 2022-04-21 by VojtechBartoska