espressif/arduino-esp32 #5887: Use arduino as a component of esp-idf in platformio

Huckies, 2021-11-14:
I'm using ESP32 Pico D4 development kit, the IDE is VScode with platformio and I wish to use the idf as well as the Arduino librarys. The system is macOS with almost all libraries installed except the libncurses-dev, I really don't know where to find it.

The situation is: Everything runs well on the default arduino environment created by platformio (the version is blank so I can't tell which exactly it is), but when I want to enable the vTaskList function, the modification on sdkconfig changes nothing on the libfreertos.a file, so I wish to use the combination of idf and arduino.

It's now quite confusing, I tried a few but not a single one works.

[This example ](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink) does compile but with my code which uses NimBLE, it returns an errer
```
In file included from lib/NimBLE/NimBLEAdvertising.cpp:25:
lib/NimBLE/NimBLEDevice.h:42:12: fatal error: esp_bt.h: No such file or directory

****************************************************************
* Looking for esp_bt.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_bt.h"
* Web  > https://platformio.org/lib/search?query=header:esp_bt.h
*
****************************************************************

 #  include "esp_bt.h"
            ^~~~~~~~~~
compilation terminated.
In file included from lib/NimBLE/NimBLECharacteristic.cpp:18:
lib/NimBLE/NimBLEDevice.h:42:12: fatal error: esp_bt.h: No such file or directory
```

I've tried many other combination of esp-idf and arduino-esp32 but they never finish the compilation, some ends up with <Arduino.h> not found and others throws pages of warnings and errors like "esp32 error: unknown type name '_lock_t'  static _lock_t rmt_driver_isr_lock;" or so.

If I create a new project using platformio and choose the espressif IoT framework and then manually revise it to
```
[env:pico32]
platform = espressif32
board = pico32
framework = arduino, espidf
```
The version of arduino-esp32 remain unknown but the idf seems to be a 4.0.1 version and not working.

I'm sorry if I raised a dumb question, but as a newbie I'm really miss now. I just want to find a solution to have my code compiled.

I used these libraries: <PubSubClient> from platformio library downloading and NimBLE from github to replace the original huge ble library.

Jason2866, 2021-11-14:
You want to use NimBLE in Arduino or as component in IDF?

Huckies, 2021-11-15:
> You want to use NimBLE in Arduino or as component in IDF?

I used to use it in Arduino but it'll be great if I can use it directly as a component in IDF

Huckies, 2021-11-15:
@Jason2866 By the way would you please tell me how to add an arduino library to the project while arduino it self is imported as a component?(forget platformio I finally made it with esp-idf v4.4 and the arduino-core v4.4)

Jason2866, 2021-11-15:
Yes, the only supported version for actual Arduino is IDF 4.4. To add it just do as for Arduino. Copy to the folder components

Huckies, 2021-11-15:
@Jason2866 It works now, thank you for being so kind!

closed 2021-11-15 by Huckies

VojtechBartoska, 2021-11-16:
Similar to #5819

Koxx3, 2022-01-29:
> @Jason2866 By the way would you please tell me how to add an arduino library to the project while arduino it self is imported as a component?(forget platformio I finally made it with esp-idf v4.4 and the arduino-core v4.4)

do you have a platformio.ini example with components folder content description ?
I can't figure out how to get : IDF + ARDUINO + Nimble components working.

Jason2866, 2022-01-29:
@Koxx3 See example https://github.com/Jason2866/platform-espressif32/tree/Arduino2021/IDF44/examples/espidf-arduino-wifiscan

Jason2866, 2022-01-29:
There is a well maintained NimBLE Arduino lib. No need to use IDF.
https://github.com/h2zero/NimBLE-Arduino

Koxx3, 2022-01-29:
i tried the weather project. it works fine, but as soon i add nimble, it fails.
i need idf to enable flash encryption 😉
so... i need the other nimble (nimble-cpp)

Huckies, 2022-01-31:
@Koxx3 Things can go different with platformio for example I have tried hard to get the idf v4.4 with arduino as component work in platformio, but ends up with failure because of the weird arduino version requirement.

You can re-compile the arduino library on your own with personal settings, espressif do offer the whole tool package. Or you can use arduino along with idf v4.4, just add it as a component to idf and write the CMakeLists manually.

happychriss, 2022-08-28:
I have been struggling with this topic for some time and finally found a solution using
platform =https://github.com/tasmota/platform-espressif32 for an ESP32.

I have summarized my finding (quite some journey) here:  
https://44-2.de/2022/08/28/best-of-two-worlds-menuconfig-options-for-esp32-arduino-and-platformio-speed-up-the-boot-time/

Would be interesting to get your feedback - I am fine to adjust when needed :-)

Jason2866, 2022-08-28:
@happychriss  Thx for using my build :-)  Actual official platformio should now work too since it uses now IDF4.4 too (which is needed for Arduino to work).

Koxx3, 2022-08-28:
i am using tasmota too. only way i found to use idf+arduino. 
excellent job @Jason2866 !
i lost hours trying to get it working without tasmota... no luck

happychriss, 2022-08-29:
@Jason2866 : First, really THANK YOU for your work. Without your build, my project would fail!!! Updated my blog-post to make this clear :-)

Like Koxx3  - I tried to get it working with official IDF4.4 - no success. 

I tried to summarize my understanding (to help out others - see my [blog](https://44-2.de/2022/08/28/best-of-two-worlds-menuconfig-options-for-esp32-arduino-and-platformio-speed-up-the-boot-time/) but honestly I am still not yet 100% clear what is the trick you are doing with your repository that makes it work. I believe its the config in the package.json???? file... just guessing...

Jason2866, 2022-08-29:
It is the config in package.json and in platform.py which frameworks (arduino / IDF) to build use for combined arduino / idf projects.
Not tried but latest release of official version looks like it should work there now.

lmurdock12, 2023-01-16:
> @Koxx3 See example https://github.com/Jason2866/platform-espressif32/tree/Arduino2021/IDF44/examples/espidf-arduino-wifiscan or this https://github.com/Jason2866/LILYGO-T5-4.7-E-Paper_Weather-Station
> 
> EDIT: Latest is https://github.com/Jason2866/platform-espressif32/tree/Tasmota/dev_apple/examples/espidf-arduino-wifiscan

Is there an updated location where all these arduino as a component within esp-idf are? I am seeing dead links all over the internet pointing to this espidf-arduino-wifiscan and I can't seem to find it anymore.

lbernstone, 2023-01-16:
It should be installed when you first build something in platformio.
https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan