espressif/arduino-esp32 #5889: 321 byte data packet receiving with UART2 Interrupt

muzahid13, 2021-11-15:
we are trying to receiving 321 byte data packet from a device over ESP32 UART2 interrupt. we didn't get more than 120 byte data. 321 byte Data length got divided into 120 byte segment. our development already finished, lastly we shucked data receiving portion. please check our Arduino IDE code below. we contacted to espressif technical team, they suggested here to raise a issue. please try to solve the issue. Thank you. 

//This is the code below
#include "driver/uart.h"
hw_timer_t * timer = NULL;
#define NUMERO_PORTA_SERIALE UART_NUM_2
#define BUF_SIZE (1024 * 2)
#define RD_BUF_SIZE (1024)
static QueueHandle_t uart2_queue;
static const char * TAG = "";                  
#define U2RXD 16
#define U2TXD 17
uart_event_t event;
size_t buffered_size;

int Second=0;
void IRAM_ATTR onTimer(){
    Second++;
     if(Second>6)
     {     
      Second=0;  
     }
        }

void setup()
{

Serial.begin(9600);

//******************start Timer Interrupt setup
/* Use 1st timer of 4 */
  /* 1 tick take 1/(80MHZ/80) = 1us so we set divider 80 and count up */
  timer = timerBegin(0, 80, true);

  /* Attach onTimer function to our timer */
  timerAttachInterrupt(timer, &onTimer, true);

  /* Set alarm to call onTimer function every second 1 tick is 1us
  => 1 second is 1000000us */
  /* Repeat the alarm (third parameter) */
  timerAlarmWrite(timer, 1000000, true);

  /* Start an alarm */
  timerAlarmEnable(timer);

//******************end Timer Interrupt setup****************************

//******************Start uart setup****************************
    uart_config_t Configurazione_UART2 = {
        .baud_rate = 9600,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    
uart_param_config(NUMERO_PORTA_SERIALE, &Configurazione_UART2);
//Firma: void esp_log_level_set(const char *tag, esp_log_level_tlevel)
esp_log_level_set(TAG, ESP_LOG_INFO);
//Firma: esp_err_tuart_set_pin(uart_port_tuart_num, int tx_io_num, int rx_io_num, int rts_io_num, int cts_io_num)
uart_set_pin(NUMERO_PORTA_SERIALE, U2TXD, U2RXD, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);

//Firma: uart_driver_install(UART_NUM_2, uart_buffer_size, uart_buffer_size, 10, &uart_queue, 0));
//  uart_driver_install(Numero_porta, RXD_BUFFER, TXD_Buffer, event queue handle and size, flags to allocate an interrupt)
uart_driver_install(NUMERO_PORTA_SERIALE, BUF_SIZE, BUF_SIZE, 20, &uart2_queue, 0);
//Create a task to handler UART event from ISR
xTaskCreate(UART_ISR_ROUTINE, "UART_ISR_ROUTINE", 2048, NULL, 12, NULL);

}

void loop() {
char test_str[7] = {0x55,0xAA,0x00,0x08,0x00,0x00,0x07};      //sending byte to IDU PCB for receiving data

if(Second>5)                //after each 5 second test_str[7] is send for receiving 321 byte data
{

uart_write_bytes(UART_NUM_2, test_str, 7);
ESP_ERROR_CHECK(uart_wait_tx_done(UART_NUM_2, 100));

  }

   }

 static void UART_ISR_ROUTINE(void *pvParameters)                   //ESP32 uart2 interrupt trigger here
{
    uart_event_t event;
    size_t buffered_size;
    bool exit_condition = false;
   
    //Infinite loop to run main bulk of task
    while (1) {
     
        //Loop will continually block (i.e. wait) on event messages from the event queue
        if(xQueueReceive(uart2_queue, (void * )&event, (portTickType)portMAX_DELAY)) {
         
            //Handle received event
            if (event.type == UART_DATA) {
 
                uint8_t UART2_data[321];                   //321 byte need to be received at atime
                int UART2_data_length = 0;                 //data length 
                ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_2, (size_t*)&UART2_data_length));
                UART2_data_length = uart_read_bytes(UART_NUM_2, UART2_data, UART2_data_length, 100);

                Serial.print("LEN= ");Serial.println(UART2_data_length);
                Serial.print("DATA= ");
                for(byte i=0; i<UART2_data_length;i++){ Serial.print(UART2_data[i],HEX); Serial.print(",");}
                Serial.println("");

          
            }
            //Handle frame error event
            else if (event.type == UART_FRAME_ERR) {
                //TODO...
            }
        }
   }
}

Arduino serial terminal output:

LEN= 120
DATA=55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,

LEN= 120
DATA=55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,5,1,1,0,1,1,F,55,AA,0,7,0,8,2,2,0,4,0,0,0,17,2D,55,AA,0,7,0,5,4,4,0,1,1,15,55,AA,0,7,0,5,5,4,0,1,2,17,55,AA,0,7,0,8,14,5,0,4,0,0,0,18,43,55,AA,0,7,0,5,65,1,0,1,0,72,55,AA,0,7,0,5,67,1,0,1,0,74,55,AA,0,7,0,5,68,1,0,1,0,75,55,AA,0,7,0,5,

LEN= 120
DATA=69,1,0,1,0,76,55,AA,0,7,0,5,6A,1,0,1,0,77,55,AA,0,7,0,5,6B,1,0,1,0,78,55,AA,0,7,0,5,6C,1,0,1,0,79,55,AA,0,7,0,8,6D,2,0,4,FF,FF,FF,DC,5A,55,AA,0,7,0,8,6E,2,0,4,0,0,0,0,82,55,AA,0,7,0,5,6F,4,0,1,1,80,55,AA,0,7,0,5,66,1,0,1,0,73,55,AA,0,7,0,8,70,2,0,4,0,0,0,3D,C1,55,AA,0,7,0,5,71,1,0,
LEN= 0
DATA= 

please help to find out the issue.

VojtechBartoska, 2021-11-15:
@SuGlider Please take a look on this issue regrading UART.

chegewara, 2021-11-15:
How is it related to arduino? This is pure esp-idf code.

SuGlider, 2021-11-15:
My two cents:

```uart_get_buffered_data_len(UART_NUM_2, (size_t*)&UART2_data_length)``` is an IDF call that will return  number of bytes available in the Rx FIFO buffer, which is limited to 128 bytes. Thus, most problably this is why your sketch is limited to 120 bytes.

muzahid13, 2021-11-15:
> As @chegewara said, this is not a really ESP32-Arduino related issue...
> 
> But, here goes my two cents:
> 
> `uart_get_buffered_data_len(UART_NUM_2, (size_t*)&UART2_data_length)` is an IDF call that will return number of bytes available in the Rx FIFO buffer, which is limited to 128 bytes. Thus, most problably this is why your sketch is limited to 120 bytes.
> 
> Given that `UART2_data_length` is then used in `uart_read_bytes(UART_NUM_2, UART2_data, UART2_data_length, 100);` It will read no more than 128 bytes each time it is called.
> 
> Good Luck!

Thanks for your comments,  understand now. Can you please help how can i read the 321 byte at a time. Which function need to use here. Please help.

SuGlider, 2021-11-16:
Try this code:

``` dart

uint8_t UART2_data[321]; //321 byte need to be received at atime 
int UART2_data_length = 0; //data length 

UART2_data_length = uart_read_bytes(UART_NUM_2, UART2_data, 321, 100); // try to read 321 at once in 100 ticks

Serial.print("LEN= ");
Serial.println(UART2_data_length); 
Serial.print("DATA= "); 

for(byte i=0; i<UART2_data_length;i++){ 
Serial.print(UART2_data[i],HEX); 
Serial.print(",");
} 

Serial.println("");

```

SuGlider, 2021-11-21:
@muzahid13 
Please let me know if this issue can be closed.

closed 2021-12-13 by SuGlider