espressif/arduino-esp32 #5893: Missing BT libraries(or not disabled properly)

Huckies, 2021-11-16:
The board is PICO-KIT-D4 V4.1, using ESP-IDF release v4.4 and arduino-core idf-release v4.4 as a component.

What I have done was moving the sdkconfig file from /components/arduino/tools/sdk to / (the root of project). My operation system is MacOS.
An additional NimBLE library "18 March 2020 - Apache NimBLE v1.3.0" is add to /components with a CMakeLists so that it's compiled. 

I do check the menuconfig and set the Bluetooth Host option to NimBLE - BLE only. To my thought the classic bluetooth should be disabled thoroughly but it shows an undefined reference to esp_gap_bt_api.h and it can be solved by copying all files in folder 
```
/components/arduino/tools/sdk/esp32/include/bt/host/bluedroid/api/include/api
```
to
```
/components/arduino/cores/esp32
```

I want to report it if it is a fault in the pre-release library, and it'll be greater if someone can point out where I made a mistake.

chegewara, 2021-11-16:
Did you try to use nimble library without changing this option?
As arduino has enabled bluedroid by default and we can use nimble library (3rd party), i think it can work also in version with as component.

Huckies, 2021-11-16:
@chegewara Can you please introduce the way of using 3rd party nimble library? Should I set the bluetooth host in menuconfig to bluedroid?

chegewara, 2021-11-16:
Just do not change bluetooth config and use the same arduino-esp32 is using.
Then use library https://github.com/h2zero/NimBLE-Arduino the same way as other components, you will have to add CMakeLists.txt or check how it is enabling nimble support and use esp-idf nimble.

I didnt test nimble with arduino as component, but it works to me on pio with arduino framework and there is a lot people using it in arduino too.

chegewara, 2021-11-16:
NimBLE is BLE only library, so no classic bluetooth and for the same reason it is smaller in usage, both, flash and ram.

Huckies, 2021-11-16:
I was comparing the nimble library carried in esp-idf and the one made by h2zero and the later seems to be thinner. What so ever, the problem is gone now, thanks for your explaination @chegewara .

closed 2021-11-16 by Huckies