espressif/arduino-esp32 #5897: ESP32 HSPI/ JTAG  conflicts with serial port

HarborDesigns, 2021-11-17:
I am using Arduino IDE (1.8.16) on Windows 10 OS. Arduino-ESP32 board library (2.0.1). I have custom board with ESP32 Wroom module. 

I am trying to interface micro SD card with ESP32 Wroom HSPI pins. I have to power cycle my board to flash code and then again power cycle to restart serial port. The program flashing works fine if the SD card is not inserted. It is something which doesn't makes sense to me as my serial port uses completely on different pins while HSPI uses different pins and protocol. 

**Observations**
* If card is inserted and I try to flash ESP over serial communication for first time it gives an error message in my Arduino IDE :

> A fatal error occurred: Timed out waiting for packet content

* I can successfully flash the code after I power cycle my board.

* After successful code flash I have to again power cycle for second time because it won't start printing on serial monitor. So after power cycle for second time my serial monitor as well as SD card read and write works.

* I can flash code and do other processes if it doesn't involves SD card or HSPI pins in a single time without power cycling. The serial monitor works fine in this case.

* I can do read/write operation of text file. I am using the [sample code from this link to test the data read and write function on SD card](https://randomnerdtutorials.com/esp32-microsd-card-arduino/). but when I try to write [JSON data on SD card using this sample code](https://www.mischianti.org/2020/01/26/manage-json-file-with-arduino-and-esp8266/) the ESP32 fails to write the JSON data and ESP32 encounters. The difference between writing a text file and JSON file is that I can successfully write a text file after power cycle but I can't do the same for JSON data. ESP32 continuously tries to reset. I have separately opened an issue with [ArduinoJSON library and you can follow the issue in this link](https://github.com/bblanchon/ArduinoJson/issues/1673). The error message for writing JSON data on SD card is shown below. I had to copy paste entire debug message as 'EspExceptionDecoder' fails to decode:

> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
SD library initialized
Delete original file if exists!
Error opening (or file not exists) /test.jso
Empty json created
Not find data array! Crete one!
Open file in write mode
Filename --> /test.jso
Start write......Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x40089be8  PS      : 0x00060930  A0      : 0x800d4ca3  A1      : 0x3ffb2470  
A2      : 0x00000002  A3      : 0xffffffff  A4      : 0x00000002  A5      : 0x00000004  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x000000f0  A9      : 0x0000004f  
A10     : 0x3ffb2480  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000042  LBEG    : 0x400d51d0  LEND    : 0x400d5234  LCOUNT  : 0x00000002  
Backtrace:0x40089be5:0x3ffb24700x400d4ca0:0x3ffb24b0 0x400d40b1:0x3ffb24d0 0x400d3313:0x3ffb2500 0x400d39fa:0x3ffb2530 0x400dfd09:0x3ffb2570 0x400dfd96:0x3ffb2590 0x400dfe00:0x3ffb25b0 0x400e03ae:0x3ffb25d0 0x400e0532:0x3ffb25f0 0x400e184f:0x3ffb2610 0x400e3466:0x3ffb2650 0x400dd1ea:0x3ffb2680 0x40086945:0x3ffb26a0 0x40085306:0x3ffb26c0 0x40085154:0x3ffb26e0 0x400851c9:0x3ffb2710 0x400d27dc:0x3ffb2730 0x400d2745:0x3ffb2750 0x400d22bd:0x3ffb2780 0x400d2492:0x3ffb27c0 0x400d5b0d:0x3ffb2820 

 
* I have observed if I pull up the MDTO/SD_CS line then the ESP32 Wroom thinks that the SD card is disabled on startup which makes sense and I can flash code when card is inserted but then when I remove the pull up resistor after forced flash then ESP32 still thinks that the CS line is low.

**Questions:**
1) Why do I have to power cycle when SD card is inserted. Please note it works fine when SD card is not there or CS is disabled. Since MDTO is on same pin as SD_CS does it impacts on ESP32 startup? I am not using JTAG in my application

2) Why does writing of text file is successful and JSON data fails?

[image]
[image]
[image]

felmue, 2021-11-17:
Hello @HarborDesigns 

AFAIK MTDI/GPIO12 is used as strapping pin which by default determines the voltage (high -> 1.8 V or low -> 3.3 V) used for the integrated flash chip.

Thanks
Felix

RoCorbera, 2021-11-17:
@HarborDesigns 
As @felmue said, there are some pins that ESP32 read or write on boot time.
This tutorial may help you on this matter: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

From this table you will see which pins are used on boot time:
[image]

HarborDesigns, 2021-11-17:
@felmue and @RoCorbera thanks I will use 10K pull down for GPIO12 on boot but do you have any explanation why for reading and writing text file on SD card [example code from RandomNerdTutorial](https://randomnerdtutorials.com/esp32-microsd-card-arduinos/) works and other [example to write JSON file fails](https://www.mischianti.org/2020/01/26/manage-json-file-with-arduino-and-esp8266/). In both cases I used HSPI pins.

HarborDesigns, 2021-11-17:
I think I resolved the issue. Part of my problem was the GPIO12 and part of it was initiation of HSPI in my code.

I basically still have to power cycle if the GPIO12 function was changed during the operation. For eg. If I have to flash code but I am using GPIO12 for SD_MISO then I have to power cycle. Upon new power cycle pull down the GPIO12 with 10K resistor and keep it low until code is flashed. After code flashing if you want to use SD_MISO you still need to power cycle for second time as the current mode was using GPIO12 as MTDI. During second power cycle you can keep the GPIO12 as floating to start the SD_MISO if you initiate it in early setup stage. 

I think in future I need to use a MOSFET or toggle button to change the functionality of GPIO12.

The second part of my problem was coding where I was not initiating and checking my SD card properly in my code. Since I am using HSPI pins and not default VSPI pins of ESP32 I had to check the card mounting status using this code:

>   SPIClass spi = SPIClass(HSPI);
>   spi.begin(SCK, MISO, MOSI, CS);
>   if (!SD.begin(CS,spi,80000000)) { //instead of what library suggest which is 'if(!SD.begin())'
>     Serial.println("Card Mount Failed");
>     return;
>   }

closed 2021-11-17 by HarborDesigns

mobizt, 2021-11-22:
If you define the **SPIClass** object that passed by reference to **SD.begin** as local variable and  you call SD member functions outside the **SPIClass** object scope, that is why you get the error **"Exception was unhandled"**.

HarborDesigns, 2021-11-24:
@mobizt Thank you for your response. That helped me figure out the issue in code.

vogelfreiheit, 2022-05-11:
```
#define SDCARD_SCK      14
#define SDCARD_MOSI     13
#define SDCARD_MISO     15
#define SDCARD_CS       2
```

I'm experiencing similar issues with a TTGO T-Beam board with the above pin configuration. I manually reset and force to OUTPUT the CS pin.

esptool fails to kick it into boot/download mode. Seems like a conflict between HSPI SPI and boot mode.