espressif/arduino-esp32 #5899: cannot use arduino libs, reports version 5.0.0 of the idf. (IDFGH-6250)

DigKleppe, 2021-11-17:
WIth the master branch cmake reports 5.0.0 instead of 4.4(?)
The latest release is 4.0.4 ( not 4.4 ? i am a little confused)
With this release  also troubles: (I have copied arduino in the idf folder)
`  CMake Error at
  /home/dig/esp/esp-idf-v4.0.4/components/arduino/CMakeLists.txt:14
  (message):

    Arduino-esp32 can be used with ESP-IDF versions between 4.4.0 and 4.4.99,
    but an older version is detected: 4.0.4.
`
Can I fool something to make arduino libs work?

igrr, 2021-11-17:
Hi @DigKleppe, ESP-IDF v4.0.4 is the last bugfix update for ESP-IDF v4.0. ESP-IDF v4.3.1 is the latest stable release of ESP-IDF at time of writing.
ESP-IDF v4.4 has not been released yet, but there is a release/v4.4 branch where this release is being prepared.
You can learn more about different ESP-IDF versions (including stable releases, pre-releases, and release branches) here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html.

Until v4.4 version is officially released, to use arduino-esp32 as an IDF component, you'll need to check out release/v4.4 branch. You can do this by following these instructions: https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/versions.html#updating-to-a-release-branch. Please use `release/v4.4` branch when running `git checkout` command.

VojtechBartoska, 2021-11-18:
same as #5819

VojtechBartoska, 2021-11-25:
@DigKleppe Did you solve your problem with using proposed versions? Thanks for update!

DigKleppe, 2021-11-26:
It works OK now, thanks.

closed 2022-04-07 by VojtechBartoska