espressif/arduino-esp32 #5900: GPIO13 not working

forfrends, 2021-11-17:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			?2.0.0?  ?2.0.1?
IDE name:						?Arduino IDE?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 7?

### Description:
Hello everyone!
I'm trying to use HSPI (GPIO 12, 13, 14, 15), but I ran into a problem, data is not transmitted and HSPI does not work at all. I tried a simple example, but it doesn't work, no data is transmitted, no clock.
At the same time, the usual blinking on these pins works. So for some reason SPI is not configured on these pins

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <SPI.h>

  #define HSPI_MISO   12
  #define HSPI_MOSI   13
  #define HSPI_SCLK   14
  #define HSPI_SS     15

static const int spiClk = 200000; // 200 KHz

SPIClass * hspi = NULL;

void setup() {
    pinMode(13, OUTPUT);
  hspi = new SPIClass(HSPI);
  
  //initialise hspi with default pins
  //SCLK = 14, MISO = 12, MOSI = 13, SS = 15
  hspi->begin();

  //set up slave select pins as outputs as the Arduino API
  //doesn't handle automatically pulling SS low
  pinMode(hspi->pinSS(), OUTPUT); //HSPI SS
}

void loop() {
  //use the SPI buses
  spiCommand(hspi, 0b11001100);
  delay(100);
}

void spiCommand(SPIClass *spi, byte data) {
  //use it as you would the regular arduino SPI API
  spi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));
  digitalWrite(spi->pinSS(), LOW); //pull SS slow to prep other end for transfer
  spi->transfer(data);
  digitalWrite(spi->pinSS(), HIGH); //pull ss high to signify end of data transfer
  spi->endTransaction();
}
```

### Debug Messages:
```
Скетч использует 224185 байт (17%) памяти устройства. Всего доступно 1310720 байт.
Глобальные переменные используют 11884 байт (3%) динамической памяти, оставляя 315796 байт для локальных переменных. Максимум: 327680 байт.
esptool.py v3.1
Serial port COM6
Connecting........_
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:aa:b5:8b:e5:e8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00046fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 474.9 kbit/s)...
Hash of data verified.
Compressed 17984 bytes to 12318...
Writing at 0x00001000... (100 %)
Wrote 17984 bytes (12318 compressed) at 0x00001000 in 0.5 seconds (effective 301.0 kbit/s)...
Hash of data verified.
Compressed 224576 bytes to 123398...
Writing at 0x00010000... (12 %)
Writing at 0x0001cfc9... (25 %)
Writing at 0x00022760... (37 %)
Writing at 0x00027a1a... (50 %)
Writing at 0x0002d012... (62 %)
Writing at 0x00037066... (75 %)
Writing at 0x0003e494... (87 %)
Writing at 0x00043c4a... (100 %)
Wrote 224576 bytes (123398 compressed) at 0x00010000 in 2.0 seconds (effective 890.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000.
```

VojtechBartoska, 2022-04-07:
Hello @HomeMade-Channel, excuse me for answering late. Is your issue still valid? If yes, are you able to retest it on v2.0.3-RC1? Thanks!

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska

SimonMalih, 2023-03-21:
Hey there using SPI.begin() with the pin numbers seemed to have worked for me.

Begin by including the SPI.h header file.

Then define HSPI pin numbers:

#define HSPI_MISO 12
#define HSPI_MOSI 13
#define HSPI_SCLK 14
#define HSPI_SS 15

in your setup function call the SPI begin function with the appropriate pins:

void setup() {
    SPI.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);      // init SPI bus
}

iharshadec, 2023-11-01:
> Hello @HomeMade-Channel, excuse me for answering late. Is your issue still valid? If yes, are you able to retest it on v2.0.3-RC1? Thanks!

Hi,

I am facing similar issue and it was fixed on 2.0.3 but it's again not working in version afterwards say latest version 2.0.14 or 2.0.11 also not working so far only version this works are 1.0.6 and 2.0.3.

do we have any fix for latest version say 2.0.11 or 2.0.14 ?