espressif/arduino-esp32 #5901: WiFiClient.h 'PGM_P' has not been declared

TrikkStar, 2021-11-17:
Attempting to use the `WiFi.h` lib and am getting a compile error with one of its dependencies.

Running ESP-IDF v4.4-dev-3569-g6a7d83af19 and using version 2.0.1 of this component.

### Debug Messages:
```
In file included from ../components/arduino-esp32/libraries/WiFi/src/wifi.h:37,
                 from ../libs/AWS_Helpers/AWS_Helpers.h:29,
                 from ../main/main.cpp:1:
../components/arduino-esp32/libraries/WiFi/src/WiFiClient.h:57:20: error: 'PGM_P' has not been declared
     size_t write_P(PGM_P buf, size_t size);
```

chegewara, 2021-11-17:
Please check if you have this file:
https://github.com/espressif/arduino-esp32/blob/2.0.1/cores/esp32/pgmspace.h

TrikkStar, 2021-11-17:
> Please check if you have this file: https://github.com/espressif/arduino-esp32/blob/2.0.1/cores/esp32/pgmspace.h

I do. Looks like if I include that, or `Arduino.h`, before `WiFi.h` it'll compile just fine.

VojtechBartoska, 2022-04-07:
@TrikkStar Can we close this as solved?

TrikkStar, 2022-04-07:
Fine by me.

closed 2022-04-07 by TrikkStar