espressif/arduino-esp32 #5902: Erase Flash options are missing from Tools menu for ESP32 Dev Module (and others?)

daxliniere, 2021-11-17:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Computer OS:                                         Windows 10

### Description:
For ESP8266 devices, we have Erase Flash options in the Tools menu. These are not present when using my ESP-WROOM-32 (38-pin dev module) and potentially other ESP32 boards.

Issue was reported [here](https://www.esp32.com/viewtopic.php?t=20621), ESP_Sprite suggested loggin a bug report here.

daxliniere, 2021-12-26:
Any news on this?

Happy holidays!

VojtechBartoska, 2022-04-11:
Hello, not yet. We will take a look on this soon and let you know about our resolution.

daxliniere, 2022-04-11:
Thank you, Vojtech! 😊

DanielMarcato, 2022-07-25:
This needs higher priority!

lbernstone, 2022-07-25:
I know it doesn't say "Erase flash", but this works.
Tools->Programmer->Esptool
Burn Bootloader

PilnyTomas, 2022-07-26:
HI, can you please give an example of which boards do have this option, and where is it located on the menu?

daxliniere, 2022-07-27:
> HI, can you please give an example of which boards do have this option, and where is it located on the menu?

It's all there in the original post, Pilny :)

lbernstone, 2022-07-27:
I think this was added in 2.0.2
https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L226-L233

daxliniere, 2022-07-30:
> I think this was added in 2.0.2 https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L226-L233

Thanks @lbernstone, I can't seem to see a 2.0.2 pre or public release online. The Arduino site has 2.0.0 rc9 plus the nightly builds (which have no version number)

PR #7043 (2022-08-10): Add erase flash option to Arduino IDE menu
## Description of Change
This PR is adding new option `Erase Flash` into Arduino IDE -> tools menu. This option can be used to erase all flash contents before uploading the sketch as it is possible for ESP8266. 
Updated docs with the new tools menu option.

[image]

## Tests scenarios
Tested on all chips.

## Related links
Closes #5902
closed 2022-08-10 by SuGlider