espressif/arduino-esp32 #5906: I2C Slave data stale

swanny101, 2021-11-18:
### Hardware:
Board:							Teyleten Robot ESP32S ESP32 ESP-WROOM-32
Core Installation version:			2.0.1
IDE name:						Arduino IDE
Flash Frequency:					N/A
PSRAM enabled:                                    N/A
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
When operating in slave mode reading from the ESP32 results in stale data. I believe I2C sends the data in the FIFO when receiving the write flag, prior to the FIFO being loaded from onRequest. Moving Wire.writes into onReceive and implementing slaveWrite within TwoWire::onReceiveService resolves the stale data.

me-no-dev, 2021-11-19:
This is limitation of the ESP32 I2C hardware. It can not clock stretch, so it can not hold the bus in onRequest for you to fill the FIFO. That is why on ESP32 you need to preload the FIFO for reading before the master actually reads. This should work as expected on ESP32-S2 and ESP32-C3 (and future chips).

closed 2021-11-19 by me-no-dev

swanny101, 2021-11-19:
So should another issue be opened against this or this one re-opened? Since onReceive does not process slaveWrite's there isn't a way to preload the FIFO for the ESP32's.

me-no-dev, 2021-11-19:
why do you think that onReceive does not process slaveWrite? Did you try to give it some time between master write and read?

swanny101, 2021-11-19:
Because I reviewed Wire.cpp. Timing wise I was using Total Wire's Aardvark test program so times between writes and reads was in the order of seconds to 10's of seconds. 

**size_t TwoWire::write(uint8_t data)**
Pushes data to txBuffer and increments txLength

**TwoWire::onReceiveService** 
Doesn't reset the length of that buffer nor does it check the length of the buffer to see if wire->slaveWrite needs to be called, nor does it call slaveWrite.

**TwoWire::onRequestService** 
Resets txLength to 0, Checks if onRequest filled the buffer then calls slaveWrite.

Proposed change:

**Origional**

```
void TwoWire::onReceiveService(uint8_t num, uint8_t* inBytes, size_t numBytes, bool stop, void * arg)
{
    TwoWire * wire = (TwoWire*)arg;
    if(!wire->user_onReceive){
        return;
    }
    for(uint8_t i = 0; i < numBytes; ++i){
        wire->rxBuffer[i] = inBytes[i];    
    }
    wire->rxIndex = 0;
    wire->rxLength = numBytes;
    wire->user_onReceive(numBytes);
}
```

**Proposed Change**
```
void TwoWire::onReceiveService(uint8_t num, uint8_t* inBytes, size_t numBytes, bool stop, void * arg)
{
    TwoWire * wire = (TwoWire*)arg;
    if(!wire->user_onReceive){
        return;
    }
    for(uint8_t i = 0; i < numBytes; ++i){
        wire->rxBuffer[i] = inBytes[i];    
    }
    wire->rxIndex = 0;
    wire->rxLength = numBytes;
	
    wire->txLength = 0;
    wire->user_onReceive(numBytes);

    if(wire->txLength){
        wire->slaveWrite((uint8_t*)wire->txBuffer, wire->txLength);
    }
}
```