espressif/arduino-esp32 #5907: I2C Slave - onRequest called for all I2C slave addresses

swanny101, 2021-11-18:
### Hardware:
Board:							Teyleten Robot ESP32S ESP32 ESP-WROOM-32
Core Installation version:			2.0.1
IDE name:						Arduino IDE
Flash Frequency:					N/A
PSRAM enabled:                                    N/A
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
When an I2C master issues a read request from a slave onRequest gets called even if the device in question was not the target slave.

me-no-dev, 2021-11-19:
can you provide minimal code that shows how you setup the Slave? What device is the master?

swanny101, 2021-11-19:
Sure. I was using a Total Phase Aardvark for my master. 

#include "Wire.h"
#include "pins_arduino.h"

#define I2C_DEV_ADDR 0x40

char regs[256] = "0";
int LED_BUILTIN = 2;

uint32_t i = 0;

char readVal = '0';
int dataReady = 0;
char counter = 0;

void onRequest()
{
  char buffer[30];
  if(dataReady == 1)
  {
    sprintf(buffer,"Sending %x",readVal);
    Serial.println(buffer);
    counter = counter + 1;
    dataReady = 0;
  }
  else
  {
    sprintf(buffer,"Called but no data to send");
    Serial.println(buffer);
  }

}

void onReceive(int len)
{
  char result[255];
  int address;
  char data;
  Serial.printf("onReceive[%d]: ", len);
  address = Wire.read();

  // Read Register
  if(len == 1)
  {
    sprintf(result,"Address 0x%x contains 0x%x",address,regs[address]);
    Serial.write(result);
    readVal = regs[address];
    dataReady = 1;
    Wire.write(readVal);
  }
  // Write Register
  else if(len == 2)
  {
    data = Wire.read();
    regs[address] = data;
    sprintf(result,"Set Address 0x%x to 0x%x",address,data);
    Serial.write(result);
    
  }
  // Length doesn't match expected
  else
  {
    while(Wire.available())
    {
      Serial.println("Error with length! ");
      sprintf(result," 0x%x",Wire.read());
      Serial.write(result);
    }
  }

  Serial.println();
}

void setup() 
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  // Setup I2C request / recieve 
  Wire.onReceive(onReceive);
  Wire.onRequest(onRequest);

  // Start I2C
  Wire.begin((uint8_t)I2C_DEV_ADDR);

  // Configure LED
  pinMode (LED_BUILTIN, OUTPUT);

}

void loop() 
{
  // Blink LED once a second for heartbeat
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000); 
}

[image]
[image]

VojtechBartoska, 2022-04-08:
Hello, can you please help with testing this on 2.0.3-rc1?

closed 2022-04-27 by me-no-dev